P0cL4bs/WiFi-Pumpkin-deprecated #2: dns spoof attack issue with scan

Zaratros, 2015-07-27:
Hello ! I was trying the new version, but I had some problem with dns spoof attack module. I couldn't scan the network properly. Here's what it says:
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/share/3vilTwinAttacker/Modules/ModuleDnsSpoof.py", line 441, in scanner_network
    for i in on_ips.values():
  File "<string>", line 2, in values
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 755, in _callmethod
    self._connect()
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 742, in _connect
    conn = self._Client(self._token.address, authkey=self._authkey)
  File "/usr/lib/python2.7/multiprocessing/connection.py", line 175, in Client
    answer_challenge(c, authkey)
  File "/usr/lib/python2.7/multiprocessing/connection.py", line 421, in answer_challenge
    message = connection.recv_bytes(256)         # reject large message
IOError: [Errno 4] Interrupted system call

mh4x0f, 2015-07-27:
Hi, this problem with scan i don't found, the line 441 return all ip scanned and after add in table. I  can to add  (try: function), but this not fix your problem. the scan not return any host scanned. however, I think that your problem is on network, pleace more details about this.

mh4x0f, 2015-07-27:
hi, I think that fixed the problem with scan IP address.

Zaratros, 2015-07-27:
Hello! The scanner works good (for now), but I got another problem. If I try for example to use windows update attack, then start dns spoofing it doesn't work.

Here's what it says:
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_template "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_template "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_WinSoftUp "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_WinSoftUp "", which already has a layout
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_WinSoftUp "", which already has a layout
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
/bin/echo: write error: Broken pipe
Starting Thread:Server-PHP
Traceback (most recent call last):
  File "/usr/share/3vilTwinAttacker/Modules/ModuleDnsSpoof.py", line 355, in Start_Attack
    open('echo 1 > /proc/sys/net/ipv4/ip_forward')
IOError: [Errno 2] No such file or directory: 'echo 1 > /proc/sys/net/ipv4/ip_forward'
Traceback (most recent call last):
  File "/usr/share/3vilTwinAttacker/Modules/ModuleDnsSpoof.py", line 355, in Start_Attack
    open('echo 1 > /proc/sys/net/ipv4/ip_forward')
IOError: [Errno 2] No such file or directory: 'echo 1 > /proc/sys/net/ipv4/ip_forward'

mh4x0f, 2015-07-28:
:D try now

closed 2015-07-28 by mh4x0f

Zaratros, 2015-07-28:
Dns spoofing works now, but it's really random. It works once in a while. I'm trying arp spoofing now, it doesn't seem to work for now It worked well in the previous versions

commit 637f4e94 (2015-07-28): fixed #2