P0cL4bs/WiFi-Pumpkin-deprecated #3: airmon-ng issues

ISeeWorld, 2015-07-31:
i use kali linux,when i click to start 3viltwin ,Pop up a dialog message box  “airmon-ng could not enable monitor mode on this device your card not support monitor mode ”,but i can use airmon-ng to start a monitor mode,i dont know how to solve this problem .anyhow,thank you for you help.

joridos, 2015-07-31:
You use Kali linux in virtualization mode(VM ware, Virtualbox)?

ISeeWorld, 2015-08-01:
Yes vmwawre7, Is that  cause this  problem?can you give me more advice ?thank for your help .

mh4x0f, 2015-08-01:
nice, send me  command output 'ifconfig' on your VM machine pleace.

ISeeWorld, 2015-08-01:
eth0      Link encap:Ethernet  HWaddr 00:0c:29:d7:b9:23  
          inet addr:192.168.1.105  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fed7:b923/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5210 errors:0 dropped:0 overruns:0 frame:0
          TX packets:156 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2215332 (2.1 MiB)  TX bytes:12566 (12.2 KiB)
          Interrupt:19 Base address:0x2000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9854 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9854 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1720379 (1.6 MiB)  TX bytes:1720379 (1.6 MiB)

wlan0     Link encap:Ethernet  HWaddr 00:f1:40:61:04:ab  
          inet addr:192.168.1.109  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2f1:40ff:fe61:4ab/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:447 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4834 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:106176 (103.6 KiB)  TX bytes:2244275 (2.1 MiB)
thank you for your help

mh4x0f, 2015-08-01:
send command output  'airmon-ng start wlan0'

ISeeWorld, 2015-08-01:
root@coolman:~# airmon-ng start wlan0
Found 4 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to kill (some of) them!

  PID Name
 2702 NetworkManager
 2830 dhclient
 4300 wpa_supplicant
 4342 dhclient

PHY	Interface	Driver		Chipset

phy0	wlan0		rt2800usb	Ralink Technology, Corp. RT2870/RT3070
		(mac80211 monitor mode vif enabled for [phy0]wlan0 on [phy0]wlan0mon)
		(mac80211 station mode vif disabled for [phy0]wlan0)
===================
thank you for your help

joridos, 2015-08-01:
What are your networks interfaces? Show the output:
`$ ifconfig -a && ip link show && netstat -i `

ISeeWorld, 2015-08-01:
eth0      Link encap:Ethernet  HWaddr 00:0c:29:d7:b9:23  
          inet addr:192.168.1.105  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fed7:b923/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:114737 errors:1 dropped:0 overruns:0 frame:0
          TX packets:7413 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:80695514 (76.9 MiB)  TX bytes:920901 (899.3 KiB)
          Interrupt:19 Base address:0x2000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:10185 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10185 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1819691 (1.7 MiB)  TX bytes:1819691 (1.7 MiB)

wlan0mon  Link encap:UNSPEC  HWaddr 00-F1-40-61-04-AB-00-00-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10106 errors:0 dropped:9633 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1494250 (1.4 MiB)  TX bytes:0 (0.0 B)

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN mode DEFAULT qlen 1000
    link/ether 00:0c:29:d7:b9:23 brd ff:ff:ff:ff:ff:ff
4: wlan0mon: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UNKNOWN mode DEFAULT qlen 1000
    link/ieee802.11/radiotap 00:f1:40:61:04:ab brd ff:ff:ff:ff:ff:ff
Kernel Interface table
Iface   MTU Met   RX-OK RX-ERR RX-DRP RX-OVR    TX-OK TX-ERR TX-DRP TX-OVR Flg
eth0       1500 0    114740      1      0 0          7414      0      0      0 BMRU
lo        65536 0     10185      0      0 0         10185      0      0      0 LRU
wlan0mon   1500 0     10109      0   9636 0             0      0      0      0 BMRU

joridos, 2015-08-01:
The problem is with [the promiscue mode in VMware](https://www.vmware.com/support/ws55/doc/ws_net_advanced_linux_vadapter_promiscuous.html)
Try this, add this in the `/etc/rc.local`:
```
    ifconfig eth0 up
    ifconfig eth0 promisc
````
And restart the network service, `/etc/init.d/network restart` or `service network restart`, and try restart the VM

ISeeWorld, 2015-08-01:
Thank you ,i will have a try!

ISeeWorld, 2015-08-01:
1 #!/bin/sh -e
  2 #
  3 # rc.local
  4 #
  5 # This script is executed at the end of each multiuser runlevel.
  6 # Make sure that the script will "exit 0" on success or any other
  7 # value on error.
  8 #
  9 # In order to enable or disable this script just change the execution
 10 # bits.
 11 #
 12 # By default this script does nothing.
 13 
 14 exit 0
 15 ifconfig eth0 up
 16 ifconfig eth0 promisc

I do as you say,but it seems does't work.

and output   "Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/share/3vilTwinAttacker/Modules/utils.py", line 105, in run
    setup_logger('airbase', './Logs/requestAP.log')
  File "/usr/share/3vilTwinAttacker/Modules/utils.py", line 230, in setup_logger
    fileHandler = logging.FileHandler(log_file, mode='a')
  File "/usr/lib/python2.7/logging/__init__.py", line 901, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 924, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 2] 没有那个文件或目录: '/usr/share/3vilTwinAttacker/Logs/requestAP.log'

SIOCSIFADDR: 没有那个设备
at0: ERROR while getting interface flags: 没有那个设备
SIOCSIFNETMASK: 没有那个设备
SIOCSIFMTU: 没有那个设备
SIOCADDRT: 网络不可达
dhpcd: no process found
dnsmasq: no process found
Starting Thread:DHCP
Starting Thread:Net-Creds
Starting Thread:Sslstrip
SSLstrip v0.9 by Moxie Marlinspike Thread::online
Traceback (most recent call last):
  File "/usr/share/3vilTwinAttacker/Core/Main.py", line 95, in run
    setup_logger('dhcp', './Logs/dhcp.log')
  File "/usr/share/3vilTwinAttacker/Modules/utils.py", line 230, in setup_logger
    fileHandler = logging.FileHandler(log_file, mode='a')
  File "/usr/lib/python2.7/logging/__init__.py", line 901, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 924, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 2] 没有那个文件或目录: '/usr/share/3vilTwinAttacker/Logs/dhcp.log'
Traceback (most recent call last):
  File "Plugins/NetCreds.py", line 1014, in <module>
    setup_logger('netcreds', './Logs/urls.log')
  File "Plugins/NetCreds.py", line 70, in setup_logger
    fileHandler = logging.FileHandler(log_file, mode='a')
  File "/usr/lib/python2.7/logging/__init__.py", line 901, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 924, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 2] No such file or directory: '/usr/share/3vilTwinAttacker/Logs/urls.log'
Traceback (most recent call last):
  File "/usr/share/3vilTwinAttacker/Core/Main.py", line 547, in StartApFake
    if not search(self.interface,check_output('ifconfig')):
  File "/usr/lib/python2.7/re.py", line 142, in search
    return _compile(pattern, flags).search(string)
  File "/usr/lib/python2.7/re.py", line 238, in _compile
    raise TypeError, "first argument must be string or compiled pattern"
TypeError: first argument must be string or compiled pattern

"

and " airmon-ng could not enable monitor mode on device wlan0mon.your card not supports monitor mode."  

thank for your help.

mh4x0f, 2015-08-01:
I fixed your problem with airmon-ng. try again :D

closed 2015-08-01 by mh4x0f