P0cL4bs/WiFi-Pumpkin-deprecated #4: No subnet declaration for at0 (no IPv4 addresses).

ISeeWorld, 2015-08-06:
I am sorry to interrupt you again,this is may be a simple question,but i google for a long time and could not solve this problem ,when i start this program shows 
"
No subnet declaration for at0 (no IPv4 addresses).
** Ignoring requests on at0.  If this is not what
   you want, please write a subnet declaration
   in your dhcpd.conf file for the network segment
   to which interface at0 is attached. **
"
iwconfig output
"
lo        no wireless extensions.

wlan1mon  IEEE 802.11bgn  Mode:Monitor  Frequency:2.457 GHz  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:on
          
eth0      no wireless extensions.

"
ifconfig output
"
eth0      Link encap:Ethernet  HWaddr 00:0c:29:d7:b9:23  
          inet addr:192.168.1.101  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fed7:b923/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:443484 errors:0 dropped:0 overruns:0 frame:0
          TX packets:257293 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:526867794 (502.4 MiB)  TX bytes:139040728 (132.5 MiB)
          Interrupt:19 Base address:0x2000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:104569 errors:0 dropped:0 overruns:0 frame:0
          TX packets:104569 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:128741477 (122.7 MiB)  TX bytes:128741477 (122.7 MiB)

wlan1mon  Link encap:UNSPEC  HWaddr E8-4E-06-20-07-68-00-00-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2909 errors:0 dropped:2908 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:860652 (840.4 KiB)  TX bytes:0 (0.0 B)

"
vi etc/network/interfaces
"
 1 # This file describes the network interfaces available on your system
  2 # and how to activate them. For more information, see interfaces(5).
  3 
  4 # The loopback network interface
  5 auto lo
  6 iface lo inet loopback
  7 
  8 # The primary network interface 
  9 #allow-hotplug eth0
 10 #iface eth0 inet dhcp

"
 i can find the device "wlan1",but a strange name "at0" appears,how could i solve this problem,can you give me some advice ?thank you very much.

mh4x0f, 2015-08-06:
hi, you need to create the AP with airbase-ng and send me ifconfig output. I need to check if the interface at0 is working, the problem is on interface at0 is not mounted.

ISeeWorld, 2015-08-07:
Thank you for your reply.i do as you say,
"root@man:~# airbase-ng  -c 9 -e sky wlan0mon
For information, no action required: Using gettimeofday() instead of /dev/rtc
12:23:36  Created tap interface at1
12:23:36  Trying to set MTU on at1 to 1500
12:23:36  Access Point with BSSID 00:F1:40:61:04:AB started.
"
"
root@man:~# ifconfig 
eth0      Link encap:Ethernet  HWaddr 00:0c:29:d7:b9:23  
          inet addr:192.168.1.105  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fed7:b923/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5201 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5533 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2889943 (2.7 MiB)  TX bytes:1334484 (1.2 MiB)
          Interrupt:19 Base address:0x2000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:7274 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7274 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1182645 (1.1 MiB)  TX bytes:1182645 (1.1 MiB)

wlan0mon  Link encap:UNSPEC  HWaddr 00-F1-40-61-04-AB-00-00-00-00-00-00-00-00-00-00  
          UP BROADCAST NOTRAILERS RUNNING PROMISC ALLMULTI  MTU:1800  Metric:1
          RX packets:14766 errors:0 dropped:840 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2429070 (2.3 MiB)  TX bytes:0 (0.0 B)
"
I could not find the at0,maybe it is the key problem.what shou i do to mount the device?Thank you very much.

mh4x0f, 2015-08-07:
I hope this commit  fix the problem, try now, thanks for issue :D

ISeeWorld, 2015-08-07:
I use the new version software,but shows
"
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
Starting Thread:Airbase-ng
2015-08-07 22:21:32,395 : ---[ Start Airbase-ng Fri Aug  7 22:21:32 2015]---
2015-08-07 22:21:32,396 : ----------------------------------------------------
Exception in thread Airbase-ng:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/share/3vilTwinAttacker/Modules/utils.py", line 110, in run
    self.process = Popen(self.cmd,stdout=PIPE,stderr=STDOUT)
  File "/usr/lib/python2.7/subprocess.py", line 679, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1259, in _execute_child
    raise child_exception
TypeError: execv() arg 2 must contain only strings
"
i dont know why.may be the this tool is not good suitable for kali.there are so many bugs.
Thank you for your help.

closed 2015-08-07 by mh4x0f