P0cL4bs/WiFi-Pumpkin-deprecated #5: Ap starting problem

heska77240, 2015-08-21:
hello i have an error message in the terminal when i start the AP:

Starting Thread:Airbase-ng
2015-08-21 17:39:31,734 : ---[ Start Airbase-ng Fri Aug 21 17:39:31 2015]---
2015-08-21 17:39:31,754 : ----------------------------------------------------
Exception in thread Airbase-ng:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/share/3vilTwinAttacker/Modules/utils.py", line 110, in run
    self.process = Popen(self.cmd,stdout=PIPE,stderr=STDOUT)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
TypeError: execv() arg 2 must contain only strings

What is the problem ?? thank you :-)

mh4x0f, 2015-08-21:
the Aircrack-ng v1.2 RC2 adds new options wlan0mon, this is problem. use the old version the aircrack-ng. it it should to work for now.
I'll try fix the problem stil. wait...

heska77240, 2015-08-21:
ok thank you for your answer where i can download a good version of aircrack ?

thank you :-)

mh4x0f, 2015-08-21:
Aircrack-ng 1.1: http://download.aircrack-ng.org/aircrack-ng-1.1.tar.gz

heska77240, 2015-08-22:
Thank you mh4x0f :-)

mh4x0f, 2015-08-23:
I fix the problem :) #6

heska77240, 2015-08-23:
hello,

there is someone to help me to make working the DNS spoof please, i make a fake ap, and i connect my android phone to the fake ap but when i make a http request on my android phone he never redirect to the fake page.... why??

i have make a video of my problem: https://youtu.be/SLHVHZGfV7U

sorry for my bad english i'm french :-)

closed 2015-08-31 by mh4x0f