P0cL4bs/WiFi-Pumpkin-deprecated #9: AWUS036H

mehranexpert, 2015-12-08:
hi dear , is it work on AWUS036H ? i not see Access point .

mh4x0f, 2015-12-08:
hi, AWUS036H not working yet AP mode, see https://wireless.wiki.kernel.org/en/users/Drivers/rtl8187

mehranexpert, 2015-12-09:
i Support on airbase-ng .

mh4x0f, 2015-12-09:
you can to change to airbase-ng on Modules->Settings, access point -> airbase-ng.

closed 2015-12-11 by mh4x0f

Highpressure, 2016-05-22:
its not selectable, is it because we need to stick to the old version of aircrack suite ?

kiang70, 2016-07-23:
it's not selectable.Ikali2.0

wikijm, 2017-07-16:
Hi everyone,

Has anyone managed to run this model with latest WiFi-Pumkin version?
I did not find "airbase-ng on Modules->Settings, access point -> airbase-ng" parameter.

wikijm, 2017-07-21:
OK, never mind, the answer is here: https://github.com/P0cL4bs/WiFi-Pumpkin/issues/94