P0cL4bs/WiFi-Pumpkin-deprecated #12: Capture device's IP and MAC address

kbroml, 2015-12-30:
Hi,

Thank you for your great work.

It works properly.

However, WiFi-Pumpkin does not show mac address, IP and the device name.

Any clues or suggestions, anything I could try...?

Kind regards, kbroml

mh4x0f, 2015-12-30:
hi, send me an print screen the terminal and tool.

kbroml, 2015-12-30:
Sorry, I forgot to specify messages output to the screen :

---%<------------------------------------------
$ ./wifi-pumpkin.py 
WARNING: No route found for IPv6 destination :: (no default route?)

(process:17707): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

(python2.7:17707): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Qt: Session management error: None of the authentication protocols specified are supported
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
Erreur : objet « radio » inconnu, essayez « nmcli help ».
dhpcd: aucun processus trouvé
dnsmasq: aucun processus trouvé
Starting Thread:hostapd
Starting Thread:DHCP
Starting Thread:sslstrip
SSLstrip v0.9 by Moxie Marlinspike Thread::online
Starting Thread:Net-Creds
2015-12-30 19:50:30,383 : ---[ Start DHCP Wed Dec 30 19:50:30 2015]---
---%<------------------------------------------
$ ps -ef | grep dhcpd | grep -v grep
root     22696  2605  0 17:37 ?        00:00:00 /usr/sbin/dhcpd -q -cf /etc/dhcp/dhcpd.conf -pf /var/run/dhcpd.pid

[image]

mh4x0f, 2015-12-30:
I can't help it, this information don't helped in nothing.please, send me print just the terminal , but connected with you device on Access point.

mh4x0f, 2015-12-30:
do you have any error in terminal when you connect you device on access point ?

kbroml, 2015-12-30:
I get no error.

My mobile device is connected to AP. The access to the Internet can be established through a gateway.

$ tail -f /var/log/syslog
Dec 30 22:47:43 localhost hostapd: wlan0: STA 30:75:12:be:f8:40 IEEE 802.11: authenticated
Dec 30 22:47:43 localhost hostapd: wlan0: STA 30:75:12:be:f8:40 IEEE 802.11: associated (aid 1)
Dec 30 22:47:43 localhost hostapd: wlan0: STA 30:75:12:be:f8:40 RADIUS: starting accounting session 56842778-00000002
[...]
Dec 30 22:47:45 localhost dhcpd: DHCPDISCOVER from 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPOFFER on 10.0.0.20 to 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPDISCOVER from 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPOFFER on 10.0.0.20 to 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPREQUEST for 10.0.0.20 (10.0.0.1) from 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPACK on 10.0.0.20 to 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPREQUEST for 10.0.0.20 (10.0.0.1) from 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 30 22:47:45 localhost dhcpd: DHCPACK on 10.0.0.20 to 30:75:12:be:f8:40 (android-bcxxxxxxxxxxxxxx) via wlan0

mh4x0f, 2015-12-30:
ok,  start the access point connected you device and check content this file -> cat WiFi-Pumpkin/Settings/dhcp/dhcpd.leases , send me output please.

kbroml, 2015-12-30:
file is empty

$ ll /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases
-rw-r--r-- 1 root root 0 déc.  30 19:50 /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases

$ /etc/init.d/isc-dhcp-server status
Status of ISC DHCP server: dhcpd is running.

 ----------------------------------------
[=] checking dependencies  
----[✔]----[+] hostapd Installed
----[✔]----[+] dhcpd Installed
----[✔]----[+] php Installed

[+] Distribution Name: Ubuntu

----------------------------------------
[=]  Install WiFi-Pumpkin 
[✔] PATH::/usr/share/WiFi-Pumpkin
[✔] binary::/usr/bin/
[✔] wifi-pumpkin installed with success
[✔] execute  wifipumpkin in terminal
[+] P0cL4bs Team CopyRight 2015
[+] Enjoy

mh4x0f, 2015-12-30:
ok man, as I thought. the information mac,ip,hostname are collected from this file dhcpd.leases, for some reason this is not happening. I recommend you try with VM (vritual Machine) on Kali.
you can to try this:
add on line 638 this line ->  if hotname == None:hostname = 'unknown'
stay like this:
633                        if hostname == None:
634                           item = leases.get_current()
635                           hostname = item[data[4]]
636                   except:
637                        hostname = 'unknown'
638                   if hotname == None:hostname = 'unknown'

kbroml, 2015-12-30:
Ok. I'll test this change. 

"or some reason this is not happening" ;-)

Thx so much

kbroml, 2015-12-31:
Hi,

No change but now I'm having problems routing out to the internet through my CPE 

Starting Thread:Net-Creds
[*] Using interface: wlan0
Traceback (most recent call last):
  File "Plugins/net-creds/net-creds.py", line 1018, in <module>
    main(parse_args())
  File "Plugins/net-creds/net-creds.py", line 1010, in main
    sniff(iface=conf.iface, prn=pkt_parser, store=0)
  File "/usr/lib/python2.7/dist-packages/scapy/sendrecv.py", line 577, in sniff
    p = s.recv(MTU)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 485, in recv
    pkt, sa_ll = self.ins.recvfrom(x)
socket.error: [Errno 100] Network is down <--- ????

$ tail -f /var/log/syslog
Dec 31 12:16:23 localhost dhcpd: DHCPOFFER on 10.0.0.20 to 30:75:12:be:f8:40  (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 31 12:16:23 localhost dhcpd: Wrote 3 leases to leases file.
Dec 31 12:16:23 localhost dhcpd: DHCPREQUEST for 10.0.0.20 (10.0.0.1) from 30:75:12:be:f8:40  (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 31 12:16:23 localhost dhcpd: DHCPACK on 10.0.0.20 to 30:75:12:be:f5:20 (android-bc7f0dda838c2734) via wlan0
Dec 31 12:16:23 localhost dhcpd: DHCPREQUEST for 10.0.0.20 (10.0.0.1) from 30:75:12:be:f8:40  (android-bcxxxxxxxxxxxxxx) via wlan0
Dec 31 12:16:23 localhost dhcpd: DHCPACK on 10.0.0.20 to 30:75:12:be:f8:40  (android-bcxxxxxxxxxxxxxx) via wlan0

$ ll /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases
-rw-r--r-- 1 root root 0 déc.  31 12:14 /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases

kbroml, 2015-12-31:
Dec 31 12:47:45 localhost dhcpd: Internet Systems Consortium DHCP Server 4.2.4
Dec 31 12:47:45 localhost dhcpd: Copyright 2004-2012 Internet Systems Consortium.
Dec 31 12:47:45 localhost dhcpd: All rights reserved.
Dec 31 12:47:45 localhost dhcpd: For info, please visit https://www.isc.org/software/dhcp/
Dec 31 12:47:45 localhost dhcpd: Can't open lease database /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases: Permission denied --
Dec 31 12:47:45 localhost kernel: [1051229.994028] type=1400 audit(1451562465.912:193): apparmor="DENIED" operation="open" profile="/usr/sbin/dhcpd" name="/usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases" pid=21098 comm="dhcpd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
Dec 31 12:47:45 localhost dhcpd:   check for failed database rewrite attempt!
Dec 31 12:47:45 localhost dhcpd: Please read the dhcpd.leases manual page if you
Dec 31 12:47:45 localhost dhcpd: don't know what to do about this.
Dec 31 12:47:46 localhost kernel: [1051230.216233] device wlan0 entered promiscuous mode

$ ll /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases
-rwxrwxrwx 1 root root 0 déc.  31 12:43 /usr/share/WiFi-Pumpkin/Settings/dhcp/dhcpd.leases*

closed 2015-12-31 by mh4x0f