P0cL4bs/WiFi-Pumpkin-deprecated #13: dont open wifi pumpin

donayxsmall25, 2016-01-05:
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/wifi-pumpkin.py", line 4, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 21, in <module>
    from Core.helpers.update import frm_githubUpdate
  File "/usr/share/WiFi-Pumpkin/Core/helpers/update.py", line 5, in <module>
    from Core.loaders.github import GithubUpdate,UrllibDownload
  File "/usr/share/WiFi-Pumpkin/Core/loaders/github.py", line 1, in <module>
    from config import Config
ImportError: No module named config

please help dont open wifi pumpkin

joridos, 2016-01-05:
What OS do you use?

```sh
$ uname -a
```

What the interfaces do you have?
```sh
$ netstat -i
```

mh4x0f, 2016-01-05:
do you need to install the dependencies ? 
```sh
$ sudo pip install -r requirements.txt
```

donayxsmall25, 2016-01-06:
mh4x0f  :  sudo: pip: command not found

joridos :   -  uname -a  : Linux donay-X450LCP 3.19.0-33-generic #38-Ubuntu SMP Fri Nov 6 18:18:12 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

*  netstat -i 
Kernel Interface table
Iface   MTU Met   RX-OK RX-ERR RX-DRP RX-OVR    TX-OK TX-ERR TX-DRP TX-OVR Flg
eth0       1500 0         0      0      0 0             0      0      0      0 BMU
lo        65536 0      1638      0      0 0          1638      0      0      0 LRU
wlan0      1500 0     11185      0      0 0          9211      0      0      0 BMRU

mh4x0f, 2016-01-06:
you need to install pip
```sh
$ sudo apt-get install pip 
```
after:
https://github.com/P0cL4bs/WiFi-Pumpkin#installation

donayxsmall25, 2016-01-07:
error  $ sudo pip install -r requirements.txt

Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-QCtO7C/Twisted/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-kv5Fo9-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-QCtO7C/Twisted

mh4x0f, 2016-01-07:
sorry, your distribution linux not support . we recommend to use Kali Linux, wifislax or Parrot.

closed 2016-01-07 by mh4x0f

ChrisShow81, 2016-08-03:
I also have this issue, "no module named config".But it's a Kali linux.
Linux kali 4.3.0-kali1-686-pae #1 SMP Debian 4.3.3-5kali4(2016-01-13) i686 GNU/Linux

what can i do next? thank you

joridos, 2016-08-04:
@ChrisShow81 did u read the [DOCS](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Installation#dependencies-softwares) !?

zacknight, 2018-05-24:
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 24, in <module>
    from core.utils import (
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 9, in <module>
    from scapy.all import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/all.py", line 10, in <module>
    from .config import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/config.py", line 11, in <module>
    from .data import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/data.py", line 184, in <module>
    TCP_SERVICES,UDP_SERVICES=load_services("/etc/services")
  File "/usr/local/lib/python2.7/dist-packages/scapy/data.py", line 109, in load_services
    f=open(filename, errors='ignore')
TypeError: 'errors' is an invalid keyword argument for this function

Rahul17navgurukul, 2018-07-13:
UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 28, in <module>
    from core.widgets.tabmodels import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/tabmodels.py", line 10, in <module>
    from core.widgets.pluginssettings import PumpkinProxySettings
  File "/usr/share/WiFi-Pumpkin/core/widgets/pluginssettings.py", line 3, in <module>
    import modules as GUI
  File "/usr/share/WiFi-Pumpkin/modules/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/modules/monitors/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/modules/monitors/credentials.py", line 2, in <module>
    from core.loaders.models.PackagesUI import *
  File "/usr/share/WiFi-Pumpkin/core/loaders/models/PackagesUI.py", line 8, in <module>
    from core.packets.network import ThARP_posion,ThSpoofAttack
  File "/usr/share/WiFi-Pumpkin/core/packets/network.py", line 1, in <module>
    import queue as Queue
ImportError: No module named queue