P0cL4bs/WiFi-Pumpkin-deprecated #14: Error Network Card : You are connected with interface wireless, try again with local connection

brandontict, 2016-01-05:
Running Kali v2 ,latest updates . 
I have two wireless cards.  Integrated chipset , and a USB Wi-Fi adapter. 

'Network controller: Intel Corporation Centrino Advanced-N 6235 ' 
'ASUSTek Computer, Inc. USB-N13 802.11n Network Adapter (rev. B1) [Realtek RTL8192CU] ' 

I have tried both wlan0 and wlan2 and still receive the following error; 
" You are connected with interface wireless, try again with local connection"  . 

On another note, Will I be able to use my USB adapter for the AP and the on board chip for injection, etc? 

I'm a little unsure. 

Looks like a great program. Beautiful interface. Thanks for the hard work!

mh4x0f, 2016-01-05:
hi, your connection with internet  comes from  wireless cards wlanX, you need to connect local connection (ethX) for work. we recommend to use VM ( Virtual Machine ) with USB Wi-Fi adapter to work  correctly.

closed 2016-01-06 by mh4x0f

TylerNakamura, 2016-01-24:
mh4x0f, could the following WiFi adapter work? is there anything that your recommend? 

http://www.amazon.com/LB1-High-Performance-Wireless-N-Computers/dp/B011ME38AG/ref=sr_1_1?s=pc&ie=UTF8&qid=1453662556&sr=1-1-spons&keywords=usb+wifi+adapter&psc=1

mh4x0f, 2016-01-25:
I don't know,execute this command on terminal:
``` sh
 iw list
```
If there is 'AP' in the list of "Supported interface modes" your device will support.
I recommended to use [Alfa AWUS036H](http://www.amazon.com/gp/product/B003YI4HRM/?tag=won00-20)

TylerNakamura, 2016-01-25:
Awesome, thank you! Great work

longjons, 2016-05-17:
Okay so far I understand it works by being connected by lan and giving the wlanx to act like ap
cool is there a way to make the app not see or ignore a specific air card as I'm running stretch on a
chrome book that doesn't have a built in lan my internal card is a Intel Wireless 7260 hence the I want to ignore and my external usb air card is a Atheros 9271 the one that works as ap great can you please help me or give me some alternative ideas I greatly appreciate it