P0cL4bs/WiFi-Pumpkin-deprecated #15: ubuntu after insalling this is what i got plz help me

koki1010, 2016-01-18:
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 3, in <module>
    import Modules as pkg
  File "/usr/share/WiFi-Pumpkin/Modules/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/Modules/monitors/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/Modules/monitors/Credentials.py", line 2, in <module>
    from Core.loaders.Stealth.PackagesUI import *
  File "/usr/share/WiFi-Pumpkin/Core/loaders/Stealth/PackagesUI.py", line 6, in <module>
    from Modules.servers.PhishingManager import frm_PhishingManager
  File "/usr/share/WiFi-Pumpkin/Modules/servers/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/Modules/servers/PhishingManager.py", line 5, in <module>
    from BeautifulSoup import BeautifulSoup
ImportError: No module named BeautifulSoup
koki@koki-desktop:~/WiFi-Pumpkin$ sudo wifi-pumpkin
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 3, in <module>
    import Modules as pkg
  File "/usr/share/WiFi-Pumpkin/Modules/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/Modules/monitors/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/Modules/monitors/Credentials.py", line 2, in <module>
    from Core.loaders.Stealth.PackagesUI import *
  File "/usr/share/WiFi-Pumpkin/Core/loaders/Stealth/PackagesUI.py", line 6, in <module>
    from Modules.servers.PhishingManager import frm_PhishingManager
  File "/usr/share/WiFi-Pumpkin/Modules/servers/__init__.py", line 7, in <module>
    module = loader.find_module(name).load_module(name)
  File "/usr/lib/python2.7/pkgutil.py", line 246, in load_module
    mod = imp.load_module(fullname, self.file, self.filename, self.etc)
  File "/usr/share/WiFi-Pumpkin/Modules/servers/PhishingManager.py", line 5, in <module>
    from BeautifulSoup import BeautifulSoup
ImportError: No module named BeautifulSoup
koki@koki-desktop:~/WiFi-Pumpkin$

mh4x0f, 2016-01-19:
```sh
$ sudo pip install BeautifulSoup
```

closed 2016-01-20 by mh4x0f

koki1010, 2016-01-22:
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 14, in <module>
    from isc_dhcp_leases.iscdhcpleases import IscDhcpLeases
ImportError:

koki1010, 2016-01-22:
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 14, in <module>
    from isc_dhcp_leases.iscdhcpleases import IscDhcpLeases
ImportError: No module named isc_dhcp_leases.iscdhcpleases

koki1010, 2016-01-22:
i install dhcp

koki1010, 2016-01-22:
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 16, in <module>
    from Core.helpers.update import frm_githubUpdate
  File "/usr/share/WiFi-Pumpkin/Core/helpers/update.py", line 4, in <module>
    from Core.loaders.master.github import GithubUpdate,UrllibDownload
  File "/usr/share/WiFi-Pumpkin/Core/loaders/master/github.py", line 1, in <module>
    from config import Config
ImportError: No module named config