P0cL4bs/WiFi-Pumpkin-deprecated #16: Error Network Card

marceloflix, 2016-01-19:
Everything seems fine except for the IPV6 route, but i don't think that's the problem.

 I run the program and i have two wireless card an internal that supports monitor mode connected to the internet and an  (Realtek Alfa AWUS036H) that supports monitor mode too, and then when i click (Start Access Point) this error appear.

Error: You are connected with interface wireless, try again with local connection.

Or is necessary that i need to be connected with an RJ45 cable to make this work?

btw, keep going the good work, this tool seems to promises a lot :+1: 

-------------------------------------------------------------------------------------------------------------------------------

WARNING: No route found for IPv6 destination :: (no default route?)
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QLayout: Attempting to add QLayout "" to frm_PhishingManager "", which already has a layout
QWidget::setMinimumSize: (/QListWidget) Negative sizes (-1,300) are not possible
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frm_Settings "", which already has a layout
QLayout::addChildLayout: layout "" already has a parent
QWidget::setLayout: Attempting to set QLayout "" on frmAbout "", which already has a layout

mh4x0f, 2016-01-19:
yes, you should is connected with RJ45 cable or run the WiFi-Pumpkin on VM (Virtual Machine) connected the interface "ethX" (Ethernet) ,connection local.

marceloflix, 2016-01-19:
Thanks!, it actually works with the virtual machine.

closed 2016-01-20 by mh4x0f