P0cL4bs/WiFi-Pumpkin-deprecated #19: with D-Link DWA-110 (USB) (Ralink RT2571W)

the-maux, 2016-01-20:
On starting the ApPoint

Starting Thread:Dns2Proxy
[*] Using interface: wlan1
Traceback (most recent call last):
  File "Plugins/net-creds/net-creds.py", line 1018, in <module>
    main(parse_args())
  File "Plugins/net-creds/net-creds.py", line 1010, in main
    sniff(iface=conf.iface, prn=pkt_parser, store=0)
  File "/usr/lib/python2.7/dist-packages/scapy/sendrecv.py", line 577, in sniff
    p = s.recv(MTU)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 485, in recv
    pkt, sa_ll = self.ins.recvfrom(x)
socket.error: [Errno 100] Network is down

the-maux, 2016-01-20:
Corrected: 
 It's just that with RT2571W you can't "ieee80211n=1" on hostapd.conf
"HT (IEEE 802.11n) in 11b mode is not allowed, disabling HT capabilities
Could not set channel for kernel driver
Interface initialization failed
wlan1: interface state UNINITIALIZED->DISABLED
wlan1: AP-DISABLED 
wlan1: Unable to setup interface.
hostapd_free_hapd_data: Interface wlan1 wasn't started"

So, just #Comment it and it's good.
You can closed.

mh4x0f, 2016-01-21:
hi, this device not have infor on drive. it probably not support AP mode. checkout 
https://wireless.wiki.kernel.org/en/users/drivers/rt73usb?s[]=ralink&s[]=rt2571w

closed 2016-01-21 by mh4x0f