P0cL4bs/WiFi-Pumpkin-deprecated #29: Access point not visible

denisprovost, 2016-02-27:
Hi,

When I run the Access Point, I get no errors but when I scan the network on my phone, this access point is not visible.
I use your tool on a virtual machine with a Hercules kali wireless card compatible with PLC interface mode.Is this a bad config me or is there a problem of the software?

Thank's in advance

mh4x0f, 2016-02-27:
hi, run the tool as root and check out the log in terminal. it probably the hostapd show the problem that is happening.

denisprovost, 2016-02-27:
Thank's for your answer,
I tried to root, same problem. I do not know if I chose the right gateway but if I had a mistake with the gateway, the access point does not would launch?

mh4x0f, 2016-02-27:
Apparently, no have problem with gateway. same if is wrong, the access point should be visible. send me a print screen or send me only output the terminal with tool running.

denisprovost, 2016-02-27:
Ok. I have attached a picture.

http://www.hostingpics.net/viewer.php?id=827108Capture.png

mh4x0f, 2016-02-27:
the problem is not the tool, not have error. Apparently, the problem can be your card adapter.

denisprovost, 2016-02-27:
I have 2 cards but my second is a ALPHA. I saw on your instructions that this card has not AP interface mode. Very strange ...

mh4x0f, 2016-02-27:
exist 2 type cards ALPHA checkout -> [adapters-support](http://elinux.org/RPi_USB_Wi-Fi_Adapters)

denisprovost, 2016-02-27:
My alpha is "Realtek Semiconductor Corp. RTL8187"

mh4x0f, 2016-02-27:
It doesn't support AP mode.

denisprovost, 2016-02-27:
Yes for the alpha card, but my hercule card work perfectly with airack :( and 'iw list' said me yes for the AP mode

mh4x0f, 2016-02-27:
the driver not support checkout -> [RTL8187](http://linuxwireless.org/en/users/Drivers/rtl8187/)

denisprovost, 2016-02-27:
For the Alpha card, ok, isn't compatible. But this picture is for Hercule Card

http://hpics.li/50a83d2

mh4x0f, 2016-02-27:
It support AP mode, I don't know what is happening.very strange, maybe you need to try scan AP with another device.

denisprovost, 2016-02-27:
Same problem with WIn7, mobile scans.

I do not understand either. Thank's for your time ;)

closed 2016-02-27 by mh4x0f

rex1997, 2016-05-17:
I am also getting same problem and my wireless card  is compatible with AP interface mode
can you resolve or if there is driver problem so how can I install driver for wifi router.
My router is Tenda

james43james, 2016-10-07:
Same problem here!
with AWUS036NH 
HOTSPOT NOT VISIBLE

Scaidenet, 2016-12-14:
**Hola, tengo un problema que cuando lo instalé me funcionó correctamente pero luego al volver a usar por segunda vez me muestra este mensaje, aqui dejo el link de la foto donde me muestra el error, que podrá ser mi error?**

https://scontent.fgye1-1.fna.fbcdn.net/v/t34.0-12/15570909_1361233057229573_1894988542_n.png?oh=e0c7a622482b8bdd08f6e10183b10da8&oe=58537D31

HinataGlo, 2017-11-04:
H!, Hotspot is visible. I connected my android but the problem is, "it is not showing any device on wifi-pumpkin", im 100% sure that i am connected to the hotspot i made
[image]