P0cL4bs/WiFi-Pumpkin-deprecated #32: wifi-pumpkin Installer   //  problem  ...

hesamer, 2016-03-22:
when install : 

root@kali:/home/kali/3vilTwinAttacker# ./installer.sh --install
[3;J

   =============================
   | wifi-pumpkin Installer|
   =============================
          Version: 0.7.3 
usage: ./installer.sh --install | --uninstall
[3;J

   =============================
   | wifi-pumpkin Installer|
   =============================
          Version: 0.7.3 
usage: ./installer.sh --install | --uninstall
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana InRelease
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana Release.gpg                                      
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana Release                                          
Err cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/non-free amd64 Packages                          
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Err cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/main amd64 Packages                              
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Err cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/contrib amd64 Packages                           
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Err cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/non-free i386 Packages                           
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Err cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/main i386 Packages                               
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Err cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/contrib i386 Packages                            
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/contrib Translation-en_US                        
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/contrib Translation-en                           
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/main Translation-en_US                           
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/main Translation-en                              
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/non-free Translation-en_US                       
Ign cdrom://[Debian GNU/Linux 2.0 _Sana_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150811-08:02] sana/non-free Translation-en                          
Ign http://download.opensuse.org  InRelease                                                                                                                  
Ign http://ftp.de.debian.org wheezy InRelease                                                                                             
Ign http://dl.google.com stable InRelease                                                                           
Hit http://ppa.launchpad.net trusty InRelease                                                                       
Get:1 http://download.opensuse.org  Release.gpg [481 B]                                                             
Hit http://ftp.de.debian.org wheezy Release.gpg                                                                                
Hit http://dl.google.com stable Release.gpg                                                   
Hit http://download.opensuse.org  Release                                                  
Ign http://download.opensuse.org  Release                                                  
Hit http://ftp.de.debian.org wheezy Release                                                        
Hit http://dl.google.com stable Release                                                                                  
Ign http://download.opensuse.org  Packages/DiffIndex                                                                      
Hit http://dl.google.com stable/main amd64 Packages                                                                 
Hit http://ppa.launchpad.net trusty/main Sources                                                                                   
Get:2 http://http.kali.org kali-rolling InRelease [24.6 kB]                                               
Hit http://ftp.de.debian.org wheezy/main amd64 Packages                                               
Hit http://ppa.launchpad.net trusty/main amd64 Packages                                                               
Hit http://ftp.de.debian.org wheezy/main i386 Packages                                                                
Hit http://ppa.launchpad.net trusty/main i386 Packages                                                          
Hit http://ftp.de.debian.org wheezy/main Translation-en                              
Hit http://ppa.launchpad.net trusty/main Translation-en                                  
Get:3 http://http.kali.org kali-rolling/main Sources [10.1 MB]                           
Ign http://download.opensuse.org  Translation-en_US                                                                                                          
Ign http://download.opensuse.org  Translation-en                                                                                                             
Hit http://download.opensuse.org  Packages                                                                                                                   
93% [3 Sources 3,925 kB/10.1 MB 39%] [Connecting to http.kali.org (192.99.200.1Ign http://http.kali.org kali-rolling/contrib Translation-en_US               
Ign http://http.kali.org kali-rolling/contrib Translation-en                  
Get:4 http://http.kali.org kali-rolling/non-free Sources [118 kB]             
Ign http://http.kali.org kali-rolling/main Translation-en_US                  
Get:5 http://http.kali.org kali-rolling/contrib Sources [62.9 kB]             
Ign http://http.kali.org kali-rolling/main Translation-en                     
Get:6 http://http.kali.org kali-rolling/main amd64 Packages [13.5 MB]         
[... 163 lines cut ...]
----------------------------------------
[+] Distribution Name: Kali

----------------------------------------
[=]  Install WiFi-Pumpkin 
[✔] wifi-pumpkin installed with success
[✔] execute  wifi-pumpkin in terminal
[+] P0cL4bs Team CopyRight 2015
[+] Enjoy

 

after  install  : 

**# root@kali:/home/kali/3vilTwinAttacker#** sudo ./wifi-pumpkin.py
Traceback (most recent call last):
  File "./wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/home/kali/3vilTwinAttacker/Core/Main.py", line 3, in <module>
    from Proxy import *
  File "/home/kali/3vilTwinAttacker/Proxy/Plugin.py", line 2, in <module>
    from Core.Utils import setup_logger
  File "/home/kali/3vilTwinAttacker/Core/Utils.py", line 15, in <module>
    import netifaces
ImportError: No module named netifaces

mh4x0f, 2016-03-23:
try this:
``` sh
$ sudo apt-get install python-dev
$ sudo pip install netifaces
```

closed 2016-03-28 by mh4x0f

djtroby, 2017-07-03:
Please if you can help to resolve this error.

root@kali:~# sudo pip install netifaces
Collecting netifaces
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 350, in _make_request
    self._validate_conn(conn)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 837, in _validate_conn
    conn.connect()
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connection.py", line 330, in connect
    cert = self.sock.getpeercert()
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/contrib/pyopenssl.py", line 324, in getpeercert
    'subjectAltName': get_subj_alt_name(x509)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/contrib/pyopenssl.py", line 171, in get_subj_alt_name
    ext = cert.extensions.get_extension_for_class(
AttributeError: 'Extensions' object has no attribute 'get_extension_for_class'