P0cL4bs/WiFi-Pumpkin-deprecated #33: install Wifi-Pumpkin error

SvipRookie, 2016-04-06:
Parrot OS

┌─[root@parrot]─[~/Desktop/WiFi-Pumpkin]
└──╼ #./wifi-pumpkin.py 
Traceback (most recent call last):
  File "./wifi-pumpkin.py", line 5, in <module>
    from Core.Main import Initialize
  File "/root/Desktop/WiFi-Pumpkin/Core/Main.py", line 21, in <module>
    from Core.Utils import (
  File "/root/Desktop/WiFi-Pumpkin/Core/Utils.py", line 15, in <module>
    import netifaces
ImportError: No module named netifaces

According to

 `mh4x0f commented 14 days ago

try this:

$ sudo apt-get install python-dev
$ sudo pip install netifaces

if you got this message

Successfully installed netifaces
Cleaning up...

now you can to run WiFi-Pumpkin.

$ sudo ./wifi-pumpkin.py`   

Unable to solve  

help me please

mh4x0f, 2016-04-14:
try this:
``` sh
easy_install netifaces
```

closed 2016-04-18 by mh4x0f

aeb5005, 2016-05-20:
This is the exact error I get when trying to install on kubuntu. When I try to install each missing module individually I still run into problems eventually. If you want more info let me know.

brandontict, 2017-05-08:
I just started receiving this same error. When attempting to install **netifaces** I receive an error, I've also tried a new virtualenv, same error 

` pip install netifaces
Collecting netifaces
  Downloading netifaces-0.10.5.tar.gz
Could not import setuptools which is required to install from a source distribution.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 387, in setup_py
    import setuptools  # noqa
  File "/usr/share/python-wheels/setuptools-32.3.1-py2.py3-none-any.whl/setuptools/__init__.py", line 10, in <module>
    from setuptools.extern.six.moves import filter, filterfalse, map
  File "/usr/share/python-wheels/setuptools-32.3.1-py2.py3-none-any.whl/setuptools/extern/__init__.py", line 1, in <module>
    from pkg_resources.extern import VendorImporter
ImportError: No module named 'pkg_resources.extern'
`

Alot of Googling has got me no where.. Any Ideas? 

Thanks for sharing this project :-D