P0cL4bs/WiFi-Pumpkin-deprecated #34: traceroute question

aeb5005, 2016-04-20:
Is it possible to run traceroute on a device connected to WiFi-Pumpkin? How does WiFi-Pumpkin alter ttl values?

mh4x0f, 2016-04-23:
hi, what's going on here is simple, to WiFi-Pumpkin sniff the packets he need configure new dhcp with address 10.0.0.1 (router). the 10.0.2.2 is your gateway(router). all packets data should pass first 10.0.0.1(Machine Attacker) ,after that  router(10.0.2.2) this is MITM attack, but  without ARP poison.

closed 2016-04-29 by mh4x0f

aeb5005, 2016-05-13:
Actually what was happening was I was running Kali in VirtualBox. The network settings were the default: https://i.imgur.com/nVwYdga.jpg

This allows TCP/IP Traffic and Ping but doesn't appear to work with UDP or ICMP. Switching to Bridged mode instead of NAT now allows traceroute through WiFi-Pumpkin.