P0cL4bs/WiFi-Pumpkin-deprecated #38: No Wi-Fi Access Point

dlynine, 2016-05-14:
kali2.0  at VM ( Virtual Machine VMware) ,wifi-pumpkin V0.7.5, i use USB network card chips is RTL8187,gui windows use default  parameter ,and no error information arise, shows starting Thread,starting Thread,
but no access point "PumpAp" can enter
thanks

RootYang, 2016-05-14:
The similar issue i meet with you,need help too,thanks!

RootYang, 2016-05-14:
@dlynine       After learning something,maybe our USB wireless not support  AP/monitor mode.So,there is no access point "PumpAp".You should execute "iw list" on terminal,if there is 'AP' in the list of "Supported interface modes" your device will support.What a pity!Try it!

RootYang, 2016-05-14:
@dlynine       After learning something,maybe our USB wireless not support  AP/monitor mode.
So,there is no access point "PumpAp".You should execute "iw list" on terminal,if there is 'AP' in the 
list of "Supported interface modes" your device will support.What a pity!Try it!

dlynine, 2016-05-14:
oh, i see,my wireless card chips RTL8187 have 3 mode ,IBSS,managed,monitor, no AP ,but at windows system can easy build a wireless WIFI point ,i think it not hardware problem,is deiver problem
https://wireless.wiki.kernel.org/en/users/Drivers

dlynine, 2016-05-15:
@RootYang i fond some interesting things, at kali 2.0.0(Linux kali 4.0.0-kali1-686-pae #1 SMP Debian 4.0.4-1~kali2 (2015-06-03) i686 GNU/Linux) i cannot build a WIFI point,but at kali 1.1.0(Linux kali 3.18.0-kali1-686-pae #1 SMP Debian 3.18.3-1~kali4 (2015-01-22) i686 GNU/Linux) can build a WIFI point(all use my wireless chips RTL8187), use command "wifi-honey" can quick test build a WIFI point or not,at 1.1.0 i can build a wifi point use wifi-honey,but use "wifi-pumpkin" canot build and arise a error "'radio' unknow, try nmcli help ", i dont know why ?

dlynine, 2016-05-15:
@RootYang i found some interesting things, at kali 2.0.0(Linux kali 4.0.0-kali1-686-pae #1 SMP Debian 4.0.4-1~kali2 (2015-06-03) i686 GNU/Linux) i cannot build a WIFI point,but at kali 1.1.0(Linux kali 3.18.0-kali1-686-pae #1 SMP Debian 3.18.3-1~kali4 (2015-01-22) i686 GNU/Linux) can build a WIFI point(all use my wireless chips RTL8187), use command "wifi-honey" can quick test build a WIFI point or not,at 1.1.0 i can build a wifi point use wifi-honey,but use "wifi-pumpkin" canot build and arise a error "'radio' unknow, try nmcli help ", i dont know why ?

psyraed, 2016-05-16:
hi cracks,

I'm using kali2.0 at VM ( Virtual Machine VMware) ,wifi-pumpkin V0.7.5, also using USB network card chips is RTL8187,and no error information arise, shows starting Thread,starting Thread,
but does not apairs any Ap on the network, i just tried with the Rj45 network . 

Also tried with Kali 1.6 , and nothing .. please any hep? 

we neer to use the network drivers with monitor mode? 
Thanks for all

closed 2016-05-18 by mh4x0f