P0cL4bs/WiFi-Pumpkin-deprecated #43: Doesn't work in VirtualBox when in bridged mode

aeb5005, 2016-05-17:
When using VirtualBox/Kali, if I use the default NAT networking mode I can run wifi-pumpkin without problem. However, I need to use ping and traceroute so I need to use bridged mode. If I switch to bridged mode I have an internet connection, but when I turn wifi-pumpkin on and connect to it I don't have an internet connection. Let me know what info you need or log files to help troubleshoot. Thanks!

aeb5005, 2016-05-19:
An update: I have no idea why, but 0.7.3 works and provides an upstream connection using virtualbox in NAT network mode, but the newest version does not. However, in the newest version I don't get error messages like I do when I start 0.7.3.

Of course if you need more details just let me know, I'm eager to help resolve this. 

Here is the output when using 0.7.3: https://gist.github.com/aeb5005/795f3ee3d9bc5ecb5b03de3f9052f073

Here is the output when using 0.7.5:

aeb5005, 2016-05-19:
Also, in the GUI under the Home tab my network adapter is wlan0. I've noticed if I go to modules -> settings and then to the advanced tab, the interface monitor defaults to wlan1. Should I be changing this to match the Network Adapter on the home tab?

mh4x0f, 2016-05-19:
do this settings of plugins and run access point. check if it work with bridged  mode.
[image]

aeb5005, 2016-05-19:
Still doesn't work. I can fine the AP and connect, but no upstream connection.
[image]

aeb5005, 2016-05-19:
I was looking through the changelog and noticed you wrote:

+- added support to start AP with wireless connection
 +    * Kali Linux (Apparently not work, need some tests) :(
 +    * support kubuntu 15.10
 +    * support wifislax - 4.11.1
 +    * support parrot - 2.0.5

I'm using Kali, I'll try some of the other recommended operating systems and see if I can get it to work. I tried to use ubuntu (not in a VM, installed to a partion on an HDD) and it had errors, I assume because I wasn't using a VM.

aeb5005, 2016-05-20:
Kubuntu: I can get it to run after manually installing some things but I do not get an upstream connection when in bridged mode. Also, it seems I get a message related to ibus-daemon. 

Terminal copy+paste:

mh4x0f, 2016-05-20:
No, you not need the VM using this systems recommended in changelog. I have an Kubuntu 15.10 ( installed to a partion on an HDD) it work same connected with my Wireless, you can see on this [screenshot](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Screenshots). also I run the tool on VMware in bridged mode and get an upstream connection normally.
[image]

 I'll try in VirtualBox comming soon

aeb5005, 2016-05-20:
I would try in VMware but I wasn't able to figure out how to get VMware to recognize the Alfa device. Could you give instructions?

mh4x0f, 2016-05-20:
it simple [check this](https://www.vmware.com/support/ws55/doc/ws_devices_usb_connect.html) or follow this image:

[image]

aeb5005, 2016-05-20:
That doesn't work for me. My Alfa device is connected to my host OS when VMware starts. I then follow the above and select connect. Now the device is connected. Oddly, it's listed as "Atheros UB91C", not the same as yours. 

If you type "iw list" or "iwconfig", do you see a connected device? That's the problem I'm having, I can apparently connect the device to VMware but it doesn't actually then allow me to connect to the internet with it or even show it's connected at all via those commands.

aeb5005, 2016-05-20:
To get the Alfa device to work in Virtualbox I have to do very specific steps:
1) restart computer
2) eject device from "connected device settings" in Windows 10 (host OS)
3) unplug
4) add the following filter to the virtualbox settings: 
[image]
5) start virtualbox
6) once the OS is loaded in the VM, plug in the device
Then it's successfully recognized by virtualbox. If I deviate from this at all, I usually have to restart the computer.

mh4x0f, 2016-05-20:
No, not need show same as me,he show  just the driver your card USB, you need connect with OS on VM.

maybe your problem is in Router, because your gateway is "10.0.0.1" the Wifi-pumpkin mount dhcp-server with same gateway "10.0.0.1"  when the  tool add the rules of iptables to share internet. this maybe is blocking the connection.

aeb5005, 2016-05-20:
Thanks, I'll check out that video.

I can get wifi-pumpkin to work using Virtualbox/NAT/0.7.3, so I looked at those settings. On the home tab, I have gateway set to 10.0.2.2. I tried switching to that and it didn't work in Virtualbox/Bridged/0.7.5

I started looking through the settings to compare and noticed this difference: 
[image]

edit:one second....

aeb5005, 2016-05-20:
As you can see in this screenshot, no matter what I change that gateway field to (10.0.0.1, 10.0.2.2, 192.168.0.1) I don't get an upstream connection like you are. 

[image]

mh4x0f, 2016-05-20:
"Maybe changing the value in the Gateway field on the home tab doesn't actually change the value?"
Yes, you not can change in the Gateway field , this settings you need change your Router.

"I started looking through the settings to compare and noticed this difference"
different this settings because on V0.7.5 the plugins not is checked. It does not interfere with anything on the internet sharing.

"Maybe you changed something ?"
I don't changed the Core of version 0.7.3 to 0.7.5

it very strange, maybe your problem is in Router really,it conflict with the WiFi-pumpkin settings.

mh4x0f, 2016-05-20:
I did some test on my Lab with same gateway and I do not get an upstream connection. I change the gateway on the tool and work for me. change this settings on Pumpkin-Settings Tab and try starting the WiFi-Pumpkin. this  should solve you problem.

[image]

after that click in "save settings".

aeb5005, 2016-05-20:
I tried that, still isn't working.

aeb5005, 2016-05-22:
I tried getting it to work in Ubuntu installed to a partition (so no VM). Here is a screenshot. I never have trouble getting the Alfa USB device to be recognized or to find the AP. 

Some unique messages: when I start the access point, i get "Error: Object 'radio' is unknown, try 'nmcli help'." and the GUI goes dark while starting. After a few minutes +, the [*] Using interface: wlan0 and below is outputted. 

Screenshot:
[image]

closed 2016-05-29 by mh4x0f

aeb5005, 2016-06-10:
for what its worth, i just cloned the repo again and i still have the same issue where the program starts fine but i never get an internet connection to work. i tried all the new dhcp settings, none worked. I'm out of ideas for troubleshooting this but would be willing to give you output/logs if you want