P0cL4bs/WiFi-Pumpkin-deprecated #50: error when invoking program outside of git clone folder

ifytbairz, 2016-05-24:
```
~> sudo wifi-pumpkin
[sudo] password for REDACTED: 
Traceback (most recent call last):
  File "/usr/bin/wifi-pumpkin", line 31, in <module>
    check_dep_pumpkin()
  File "/home/REDACTED/Development/gitClones/WiFi-Pumpkin/Core/loaders/checker/check_depen.py", line 28, in check_dep_pumpkin
    copy('Settings/source.tar.gz','Templates/')
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: 'Settings/source.tar.gz'
exit 1

```

changing "Settings/source.tar.gz" and "Templates/" in check_depen.py, line 28

`copy('Settings/source.tar.gz','Templates/')`

to absolute paths could be possible fix .

mh4x0f, 2016-05-24:
thank's for report :D

ifytbairz, 2016-05-24:
thank you! issue resolved!

closed 2016-05-24 by ifytbairz