P0cL4bs/WiFi-Pumpkin-deprecated #52: Anybody made it to work?

GGGunrunner, 2016-05-29:
Tried on:
Kali 2.x
Kubuntu (16.x)
Parrot OS (latest)

Virtualbox, Alfa AWUS036H.

The installer completes without errors. Had to manually install hostapd.
Then, starting the ap.

┌─[root@parrot]─[~/WiFi-Pumpkin]
└──╼ #./wifi-pumpkin.py 
Qt: Session management error: None of the authentication protocols specified are supported
Loading GUI...
WiFi-Pumpkin Running!
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Starting Thread:hostapd
Starting Thread:DHCP
Starting Thread:sslstrip

sslstrip 0.9 by Moxie Marlinspike running...
sergio-proxy v0.2.1 online
Starting Thread:Dns2Proxy
Starting Thread:Net-Creds
=====

And this is it. It never appears online. Says its "ON", but none of the clients can see it.
What can be fixed?

sscarduzio, 2016-05-29:
I managed to install the missing SSL stuff etc, now I can see the AP with my phone, but can't obtain an IP..

sscarduzio, 2016-05-29:
Rebooted, made it work. Saw my bank's passwords on the net creds... Mind: blown.

mh4x0f, 2016-05-29:
@GGGunrunner if you not see the AP, because your hardware( Alfa AWUS036H) is the problem. please read the [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Installation#hardware)

closed 2016-05-29 by mh4x0f