P0cL4bs/WiFi-Pumpkin-deprecated #53: BadDrawable (invalid Pixmap or Window parameter)

cognoscibilis, 2016-06-01:
Multiple errors on start, GUI not working
[image]

```
$uname -a
Linux v3 4.4.0-kali1-amd64 #1 SMP Debian 4.4.6-1kali1 (2016-03-18) x86_64 GNU/Linux
```
`
 sudo ./wifi-pumpkin.py 
`
> Loading GUI...
> WiFi-Pumpkin Running!
> X Error: BadAccess (attempt to access private resource denied) 10
>   Extension:    130 (MIT-SHM)
>   Minor opcode: 1 (X_ShmAttach)
>   Resource id:  0x15b
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: known incorrect sRGB profile
> X Error: BadShmSeg (invalid shared segment parameter) 128
>   Extension:    130 (MIT-SHM)
>   Minor opcode: 5 (X_ShmCreatePixmap)
>   Resource id:  0xd4
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x1e00011

mh4x0f, 2016-06-01:
add this line `QT_X11_NO_MITSHM=1`  in ~/.profile file. or  (ie: system env var).
``` sh
export QT_X11_NO_MITSHM=1
```

closed 2016-06-16 by mh4x0f

ricardolink, 2017-05-21:
i have the same problem here... can u help me out ?

mh4x0f, 2017-05-22:
Based on a google search this looks like a Qt/KDE problem. Alternatively, if possible, try a different OS/distro.

becare, 2017-06-16:
sudo nano /etc/environment

Add this line
QT_X11_NO_MITSHM=1

Enjoy!

ruggboy, 2017-06-28:
becare.... You are Awesome, I'm sure you know this.

MaximeSoulie, 2017-07-27:
> sudo nano /etc/environment

Add this line
QT_X11_NO_MITSHM=1

Enjoy!
> 

Oh yes ! Thanks !

999jay999, 2017-07-30:
Hi guys, this didn't seem to work for me.

Any further advice? Also, I don't know if this matters or not, the environment file didn't have anything in it at all until I placed the code you guys had mentioned. Should it have had some sort of coding in the file?

After updating it with the code `QT_X11_NO_MITSHM=1` I continued to get the same problem, nothing had changed.

Thanks in advance.

MaximeSoulie, 2017-07-31:
@999jay999, maybe you are missing some prerequisites for Qt. What is your error message and which distro do you use?

999jay999, 2017-08-01:
I'm using Kali 2017.1 on Debian 8.x 64 bit in VMware. I get the following when attempting to run wifi-pumpkin:

> Loading GUI...
> WiFi-Pumpkin Running!
> X Error: BadAccess (attempt to access private resource denied) 10
>   Extension:    131 (MIT-SHM)
>   Minor opcode: 1 (X_ShmAttach)
>   Resource id:  0x12e
> X Error: BadShmSeg (invalid shared segment parameter) 128
>   Extension:    131 (MIT-SHM)
>   Minor opcode: 5 (X_ShmCreatePixmap)
>   Resource id:  0x1200019
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x120001a

The environment file only contains `QT_X11_NO_MITSHM=1`

I have now created a new Kali system and it seems to be working completely fine on there. However, it would still be worth knowing how to fix this issue; I have all the main programs installed in the 1st Kali VM I created.

Thanks

3bu1, 2018-01-16:
I am using kali linux, even i was facing this issue after lot of research i got to know kali doesnt allow any app run as root. I tried switching to a user and execute the program "woooolaaaa" its working perfectly.

macmakran, 2018-11-14:
thanku bro

brunomarks7, 2019-01-23:
> sudo nano /etc/environment
> 
> Add this line
> QT_X11_NO_MITSHM=1
> 
> Enjoy!

thanks! works on System76/PopOS version

JohnnyDeeee, 2019-05-20:
The provided solutions are not working for me...

I'm connecting to my Raspberry Pi (Kali Linux) through VNC.
The weird thing is when i tried @3bu1 solution the program always asked for sudo rights, so i'm not able to run it as a non-sudo user...

@mh4x0f is there any other fix/workaround for this issue?

marianantononofrei, 2019-06-26:
> sudo nano /etc/environment
> 
> Add this line
> QT_X11_NO_MITSHM=1
> 
> Enjoy!

It actually worked!

ftmzhrsckldr, 2019-07-23:
I logged in as root and got this error. Without root, the problem is solved.

WinterSoldier13, 2020-01-04:
> Hi guys, this didn't seem to work for me.
>
> Any further advice? Also, I don't know if this matters or not, the environment file didn't have anything in it at all until I placed the code you guys had [...]

Try this:
sudo QT_X11_NO_MITSHM=1 python2 wifi-pumpkin.py