P0cL4bs/WiFi-Pumpkin-deprecated #54: an connection with wlan0 has been detected : devise or resource busy

nassim2800, 2016-06-09:
i have problem with WiFi-Pumpkin
i use kali linux 2.0 in vmawre workation 11.0.0 when i satrt server and i want to hack pc with my connection and lick start access point and show always this erreur 
an connection with wlan0 has been detected : devise or resource busy
wlan0     IEEE 802.11bgn  ESSID:"DJAWEB_3B69"
i have wirless adaper tplink-wn722 Atheros 
[image]

aeb5005, 2016-06-09:
Try to disable the wifi, then start wifi-pumpkin.

nassim2800, 2016-06-10:
i am sorry i was work in windows update adapter then i click access point this is problem when i stopped the widows update attack it work correctly
but i have another problem with windows update attack when i click server start it start normaly but when i go google chrome  i don't found the fake page only i open the ip 192.168.15:80 
what is the solution i want to shhow the fake page in all websites

closed 2016-06-10 by mh4x0f

nassim2800, 2016-06-10:
the problem is corect but i have another problem the dns spoof not work because when i active dns with update fake page i don't show the fake page in browser i try wlan0 and eth0
what is the problem please 
[image]

mh4x0f, 2016-06-10:
please, you need to create another issue for this question only dns spoof.