P0cL4bs/WiFi-Pumpkin-deprecated #55: problem with dns spoof

nassim2800, 2016-06-10:
i have problem the dns spoof not work because when i active dns with update fake page i don't show the fake page in browser i try wlan0 and eth0
what is the problem please 
[image]

mh4x0f, 2016-06-12:
do you try to start the AP (Access point) and run DNS spoof correct?

nassim2800, 2016-06-12:
no but dns spoof work withowt AP i showed your video tutorial 
but when i active AP What is the benefit dns spoof 
what is the problem i don't understand i try in wi xp and 8 don't success
can you help me

mh4x0f, 2016-06-12:
i don't understand what you want, send me output the terminal.

closed 2016-07-30 by mh4x0f