P0cL4bs/WiFi-Pumpkin-deprecated #60: libpng warning: iCCP: known incorrect sRGB profile Error

Disc0nect, 2016-07-02:
I ran through the install several times; however, when I launch the application I receive the following error "libpng warning: iCCP: known incorrect sRGB profile."

[image]

After launching the application, I receive and error and the AP does not broadcast.

[image]

Hardware /Software:
Windows 10 box running Virtual box
Kali 2016 image
All updates and upgrades applied
Wifi Antenna -  AWUS036NH

mh4x0f, 2016-07-02:
it's a warning and not an error, just ignore. the kali 2016 not have support the tool, some dependency not work fine.

Disc0nect, 2016-07-02:
The error prevents the application from broadcasting an SSID. Any recommendations?

mh4x0f, 2016-07-02:
no, not error that prevented. use kali 2.0 or all OS listed on wiki instalation

mh4x0f, 2016-07-05:
thanks for report :D

Disc0nect, 2016-07-06:
Anytime, it's really a great tool.  I tested the changes on both Kali Sana & rolling.  The new changes fixed the crash issue on both distros.  Everything appears to work correctly now, with the exception or the credential portion.  This section no longer grabs credentials, nor generates an error to notate.

Thanks again for the quick fixes.. it's much appreciated.

closed 2016-07-27 by mh4x0f