P0cL4bs/WiFi-Pumpkin-deprecated #62: Having problem access the spoofed pages on the victim PC

Cosmic4x2, 2016-07-14:
Hello.. I have been trying to get you program working for a while its seem so be an awesome program

I start the sever it creates the Fake AP I connect to the AP but it does not route to any pages. I have set to windows update spoof with payload  but the only way I can get to it is by 10.0.0.1 in the browser. Please let 
me know if I am doing something wrong or how to fix this..

Thx

mh4x0f, 2016-07-16:
hi, You need to run  Fake AP without plugin Proxy server, to work. In latest release was been added in plugins an option "No Proxy". for create the Fake AP with DNS or ARP spoof this option "No Proxy" should be marked.

mh4x0f, 2016-07-17:
check this screenshot:
[image]

Cosmic4x2, 2016-07-17:
Hello I do not have that option. Is there an updated version?.... Also I
appreciate the help let me know where I can send  donations....

[image: Inline image 1]

Cosmic4x2, 2016-07-17:
AHHH Its Works.... Fantastic.... Let me know about the donations.... Much
Thanks Good Work..............

mh4x0f, 2016-07-17:
about the donations, I'll add this options coming soon... thanks for report.

Cosmic4x2, 2016-07-18:
Cool... Please keep me posted on any other projects you may have...

Thx

mh4x0f, 2016-07-19:
added donate :D https://github.com/P0cL4bs/WiFi-Pumpkin#donate

Cosmic4x2, 2016-07-19:
can you add a link for paypal...

Cosmic4x2, 2016-07-19:
Hey found the link sent you a donation... Keep me in mind for any up coming
projects...

Thx

mh4x0f, 2016-07-20:
thanks for donation :+1:

closed 2016-07-22 by mh4x0f

Cosmic4x2, 2016-07-23:
Hello On Pumpkin when target PC is connected to the rogue AP is it possible
to route all traffic back to the spoofed page or is it possible to
spoof only 1 page but with target PC still able to browse the web.

Thx