P0cL4bs/WiFi-Pumpkin-deprecated #65: DNS Spoof with Rougue AP

Cosmic4x2, 2016-07-25:
Is it possible to route all HTTP to the spoofed update page when AP is enabled.

mh4x0f, 2016-07-25:
yes, when you start dns spoof attack with AP is enabled all client connected the AP are redirected to page spoofed. for this you need to clear all hosts added on DNS spoof Module.
check this screenshot:
[image]

Cosmic4x2, 2016-07-26:
Cool it works...

Thx

mh4x0f, 2016-07-26:
@Cosmic4x2  I'll fix Dns spoof module coming soon for as you said "possible to
spoof only 1 page but with target PC still able to browse the web". 
thanks for suggestion

mh4x0f, 2016-07-30:
:D check the video demo
https://www.youtube.com/watch?v=SDBVgqYw0OE

Cosmic4x2, 2016-07-31:
Nice working good now...

Thx

closed 2016-07-31 by mh4x0f

CamelToeJoe, 2016-09-20:
When I add hosts it will redirect them to the spoof page.  However when I clear all hosts, then nothing gets redirected to the spoof page.

I am trying to redirect all traffic to a certain page.  

Thanks

Cosmic4x2, 2016-09-20:
make sure you are choosing no Proxy on plugins

CamelToeJoe, 2016-09-20:
I did.  I can spoof all the  single pages I want.   Works fine.   I just want to redirect it all.

mh4x0f, 2016-09-21:
@CamelToeJoe create an new for this question. I'll check coming soon...

mh4x0f, 2016-09-21:
@CamelToeJoe It is no longer necessary create new issue. I fixed this bug... thanks

Cosmic4x2, 2016-09-21:
Great how did you fix it?

mh4x0f, 2016-09-21:
@Cosmic4x2 I did somes test in my lab, for redirect all HTTP page now you need "clear all" hosts in module DNS spoof as I showed in [screenshot](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/65#issuecomment-235054677)  above.