P0cL4bs/WiFi-Pumpkin-deprecated #66: Beef Injection

Cosmic4x2, 2016-07-26:
Hello again I hate to keep bugging you but I have another question. When trying to use beef hook injection it does not seem to be working for me. I have  sergio-proxy selected set hook url http://192.168.1.87:3000/hook.js  it shows that I have several injection but is not showing up in beef. I have tried using http://10.0.0.1:3000/hook.js did not work either strangest thing is it did work a few times  I followed the same steps as when I was able to connect. Also from time to time it seems to kill my apache.

mh4x0f, 2016-07-27:
I'll do some test for this bug coming soon

mh4x0f, 2016-07-27:
I did not have problem with beef hook, all works fine. maybe, the website that you are testing is not to be injected. check this 
[image]

Cosmic4x2, 2016-07-27:
Actually I think the problem is I'm getting a two many files open error Beef is connecting but after the target machine browses a couple of pages I get that error then cannot browse anymore machine still shows connected  to Rouge AP but it looses web and beef connection.

mh4x0f, 2016-07-27:
read this paper about this error "Too many open files".
[https://easyengine.io/tutorials/linux/increase-open-files-limit/](url)

Cosmic4x2, 2016-07-28:
Yea that did work still having a problem hooking beef may be a problem on my end.. I will check it out. Thanks for the quick response.

closed 2016-08-01 by mh4x0f