P0cL4bs/WiFi-Pumpkin-deprecated #68: wifi-pumpkin Installer kali problem

linuggx, 2016-07-30:
I have a problem when i tried to start wifi-pumpkin:

root@kali:~/WiFi-Pumpkin# sudo ./wifi-pumpkin.py
Traceback (most recent call last):
  File "./wifi-pumpkin.py", line 39, in <module>
    from Core.Main import Initialize
  File "/root/WiFi-Pumpkin/Core/Main.py", line 28, in <module>
    from Core.widgets.TabModels import (
  File "/root/WiFi-Pumpkin/Core/widgets/TabModels.py", line 9, in <module>
    from Core.widgets.docks.DockMonitor import dockAreaAPI
  File "/root/WiFi-Pumpkin/Core/widgets/docks/DockMonitor.py", line 2, in <module>
    from pygtail import Pygtail
ImportError: No module named pygtail

Thanks.

linuggx, 2016-07-30:
pip install pygtail the solution. sorry and thanks again.

mh4x0f, 2016-07-30:
thank's , this module is not requirements anymore.

closed 2016-07-31 by mh4x0f

Viveksgautam, 2018-01-12:
Traceback (most recent call last):
  File "./wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/root/Documents/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/root/Documents/WiFi-Pumpkin/core/main.py", line 55, in <module>
    from netfilterqueue import NetfilterQueue
ImportError: No module named netfilterqueue

Viveksgautam, 2018-01-12:
Collecting netfilterqueue
  Using cached NetfilterQueue-0.8.1.tar.gz
Building wheels for collected packages: netfilterqueue
  Running setup.py bdist_wheel for netfilterqueue ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VV1f3e/netfilterqueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp0vswKjpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_ext
  building 'netfilterqueue' extension
  creating build
  creating build/temp.linux-x86_64-2.7
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-HVkOs2/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c netfilterqueue.c -o build/temp.linux-x86_64-2.7/netfilterqueue.o
  netfilterqueue.c:437:42: fatal error: libnfnetlink/linux_nfnetlink.h: No such file or directory
   #include "libnfnetlink/linux_nfnetlink.h"
                                            ^
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for netfilterqueue
  Running setup.py clean for netfilterqueue
Failed to build netfilterqueue
Installing collected packages: netfilterqueue
  Running setup.py install for netfilterqueue ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VV1f3e/netfilterqueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-OWfDiJ-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'netfilterqueue' extension
    creating build
    creating build/temp.linux-x86_64-2.7
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-HVkOs2/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c netfilterqueue.c -o build/temp.linux-x86_64-2.7/netfilterqueue.o
    netfilterqueue.c:437:42: fatal error: libnfnetlink/linux_nfnetlink.h: No such file or directory
     #include "libnfnetlink/linux_nfnetlink.h"
                                              ^
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VV1f3e/netfilterqueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-OWfDiJ-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-VV1f3e/netfilterqueue/
root@vivek:~/Documents/WiFi-Pumpkin#