P0cL4bs/WiFi-Pumpkin-deprecated #69: 2 wifi wireless adabter

bayp1p0, 2016-07-30:
I wanna use 2 wireless adabter because cable network is not usefull how can ı do this?

mh4x0f, 2016-07-30:
just use OS with support to exclude the card adapter on NetworkManager ( except Kali ) some tested in this line.
https://github.com/P0cL4bs/WiFi-Pumpkin/blob/master/CHANGELOG#L27

bayp1p0, 2016-07-30:
I am using parrot 3. I wıll try parrot 2.0.5 ıf work ı will use 2.0.5 versıon just for wifi pumpkın. But Please, add parrot 3 versıon for support

mh4x0f, 2016-07-30:
I'll do some test with parrot 3 coming soon

bayp1p0, 2016-07-30:
Ovvv good news. Thanks bro ı am waıtıng god bless you :)

mh4x0f, 2016-07-31:
try now... :D I think this  solved  the problem.

bayp1p0, 2016-07-31:
I thınk you are best best :) ı tryed but my system parrot 3.0 and it ıs work but not create AP, ı am upgrading my system now. I am sure work when system upgrade. Thanks bro again ıf work ı wrıte here

bayp1p0, 2016-07-31:
Unfortunatly everythınk ıs look good but no create ap.? :(

mh4x0f, 2016-07-31:
if everythınk ıs look good but no create ap, your problem is not the tool, probably your USB card. thanks

closed 2016-07-31 by mh4x0f

bayp1p0, 2016-07-31:
Whıch usb card support?  I can buy.

mh4x0f, 2016-08-01:
https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Installation#hardware

brandontict, 2016-08-01:
I look

bayp1p0, 2016-08-19:
https://www.amazon.com/Alfa-AWUS036NH-Wireless-Long-Range-Screw-On/dp/B003YIFHJY

I bought this adabter but still not working. Cant create fakeap :(

mh4x0f, 2016-08-20:
try this solution for chipset RT3070 problem...
installing rt2x00lib v0.35 in wheezy. Here is what I did:
In /etc/apt/sources.list, added
deb http://ftp.us.debian.org/debian wheezy main contrib non-free
and commented out other components, and then ran
aptitude update
aptitude install firmware-ralink 

(refer)[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=6292]

bayp1p0, 2016-08-20:
http://i.hizliresim.com/kEj9pD.jpg as you can see everythıng look work

http://i.hizliresim.com/5Y8azz.jpg
But no create ap. 
I am wondering that why wifi card not change monıtor mode. (wlan1 to wlan1mon). If you can add skill maybe work for every wifi adabter

mh4x0f, 2016-08-20:
it not is necessary change card to monıtor mode for mount AP with  hostapd. send me output the file Logs/AccessPoint/hostapd.log.

bayp1p0, 2016-08-20:
Hostapd.log folder is empty there is not log :( 
I was buıld my parrot 3.1 system but again error 
lf ı will change wlan0 to eth0 is it work?

bayp1p0, 2016-08-20:
And maybe ı can use aırbase ng maybe work but ı cant choıse aırbase

mh4x0f, 2016-08-20:
```
And maybe ı can use aırbase ng maybe work but ı cant choıse aırbase.
```
look, if not is create  AP the problem is hostapd,I removed the airbase,not it possible to use anymore. 
```
lf ı will change wlan0 to eth0 is it work?
```
I don't believe that the problem be this , I don't believe that the problem be this, but you can try. you should to make some test with hostapd and check if create AP, if it work, I 'll check the problem.

bayp1p0, 2016-08-20:
I can use sıngle hostapd.  Hostapd work when ı use manuel but wifi-pumpkın cant create ap

bayp1p0, 2016-08-20:
I will take a video about this problem and send you after 1 -2 days

mh4x0f, 2016-08-20:
ok, when you record the video, set the change the screenshot that i sent you above.

mh4x0f, 2016-08-20:
send me the file hostapd.conf that you are using with Hostapd manual.

bayp1p0, 2016-08-20:
" interface=wlan0mon 
driver=nl80211
ssid=FakeAp
Channel=11"
This is my hostapd.conf

mh4x0f, 2016-08-20:
ok, if hostapd work, you are correct the problem can be the connection wlan0 (internet). check if work with eth0. maybe the Parrot not is exclude the interface of NetworkManager.conf this same error  is occuring on Kali linux.

bayp1p0, 2016-09-07:
Error error continues. I buıld parrot 3.1 again but not create AP.  Please take a video about that wifi-pumpkın with 2 wifi adabter :(

mh4x0f, 2016-09-07:
it not necessary, because for using with 2 wifi adapter you need  just using an system that support exclude card adapter in NetworkManager.conf , I tested this option only in  OS:Ubuntu 15.10 KDE because this is my desktop. I can't test this options in another OS in the moment. thanks

mh4x0f, 2016-09-08:
@bayp1p0 you can try to do this, change the NetworkManger.conf to ignore card adapter and restart the network-manager process. follow below...
* open the terminal in parrot and edit this file 

> nano /etc/NetworkManager/NetworkManager.conf

``` sh
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false
``` 
* now open another terminal and get MAC address of the card adapter with command `ifconfig`. after that, you have to add the new key ([keyfile]) in file NetworkManager.conf as below.
``` sh
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=mac:00:19:e0:57:86:af # your mac address card adpter (for using Wifi-pumpkin)
```
You can restart network-manager by doing (in the terminal):
`sudo service network-manager restart`

This configuration makes NetworkManager oblivious to the existence of the wireless card, but still allows me to control it using other methods.

now, you have to connect with default card wireles and  start WiFi-Pumpkin select the adapter (my case 00:19:e0:57:86:af) that you ignore the configuration above.

OBS: **when you stop the access point in WIFI-Pumpkin and after that, the configuration that you did in NetworkManager file,it will be automatically removed**

 if does work try again configuration and run hostapd manually and check if mount AP send me if work...

bayp1p0, 2016-09-08:
Yeaap thanks bro I get some error error but I fıxed that and finally work. 
Thank you again :)

bayp1p0, 2016-09-08:
But not persıst how can ı persıst that

mh4x0f, 2016-09-08:
I'll add one option for disable  **automatically remove  key in NetworkManager.conf** configuration coming soon...

bayp1p0, 2016-09-08:
I added startup applıcatıon my conf :) ı am waıtıng update. Wifi-pumpkın is best wifi tool thanks for this tool bro

mh4x0f, 2016-09-08:
now you just need check this option "Ignore USB Wi-Fi Adapter permanently"
[image]

CamelToeJoe, 2016-09-22:
I have tried the above mention fix for using two wireless cards.  I cannot seem to get it working.  Does changing the network.conf filework instead.  I have an internal wireless and an external USB dongle.  I have tried connecting with both ways and using either as the Fake AP with the other connected to the internet and get errors both ways.  One will say the card is busy and the other way it will say you are connected to a wireless interface, please connect with a local area connection.

It would be nice to get this working so it could be portable.

mh4x0f, 2016-09-23:
@CamelToeJoe you probably is using kali linux, this settings only for Parrot OS. using  Kali linux it will not work with 2 card, I've blocked in code, because on Kali Linux need some tests. you can this remove this lines in code Main.py and check if does work. 
https://github.com/P0cL4bs/WiFi-Pumpkin/blob/master/Core/Main.py#L932-L935

CamelToeJoe, 2016-09-23:
Yes Kali it is.... I will play around with this today or tomorrow.

Thanks

jeretc, 2016-11-08:
Solved it by installing wicd manager