P0cL4bs/WiFi-Pumpkin-deprecated #70: Create a Debian/Ubuntu package

joridos, 2016-08-01:
Create a Debian/Ubuntu package for easy install in Debian based systems.

joridos, 2016-11-29:
Having some problems with the external Python module to generate the deb installer

TimSchumi, 2016-12-05:
I'll read a bit into the process of creating a .deb file and see if I can help

joridos, 2016-12-05:
Yes, I'm having some problems with the python libraries that are not allowed to install them from the apt repository.

TimSchumi, 2016-12-06:
See this: http://askubuntu.com/questions/62534/adding-post-deb-install-scripts
In this we can input the pip stuff and all the stuff we need to add.
I also recommend that we change WiFi-Pumpkin to wifi-pumpkin at the installation folder, because this follows the Debian/Ubuntu packaging recommendations and matches the program name

TimSchumi, 2016-12-06:
I made a first version of a "Auto-DEB" Script: http://pastebin.com/k1B43DRn
I need to fix dependency checking though...

TimSchumi, 2016-12-08:
Should I add the make_deb.sh file to the .travis-ci.yml file?

mh4x0f, 2016-12-10:
@TimSchumi added you contribution in about WiFi-Pumpkin thanks again :D

closed 2016-12-10 by mh4x0f

TimSchumi, 2016-12-10:
@mh4x0f Why did you add the i386 and amd64 packages? I thought the "all" package was good for all architectures...

mh4x0f, 2016-12-10:
@TimSchumi only by choice.

TimSchumi, 2016-12-10:
@mh4x0f I'll stick with the 'all' package, as I don't really get the difference between seperate amd64 and i386 packages. I also recommend that if you build a deb package, that you change the Version Number in the Script for one time instead of renaming the final package. So the internal dpkg version tracking is working/you can identify the versions.

mh4x0f, 2016-12-10:
@TimSchumi no problem, it necessary seperate the Arch amd64 and i386 packages, because need to download correct packages in  "Depends:" and add arguments it not good ideia because the make_deb.sh only used for generate .deb. if the user want .deb he should make download on  Release.

TimSchumi, 2016-12-10:
@mh4x0f When I tested it, the all package requested and installed the correct dependencies every time. And the Arguments should make it easier for you when building, because you else had to change Architecture and Version in the Script every time.