P0cL4bs/WiFi-Pumpkin-deprecated #71: Device or resource busy

linuggx, 2016-08-01:
In reference to the issue #54 

Hello, I have the same problem of the issue 54: 
I use Kali Linux and when I click start access point WiFi-Pumpkin shows me always this error:
an connection with wlan0 has been detected : device or resource busy
wlan0 IEEE 802.11bgn ESSID:"name_wifi".
I have tried to do what he said @aeb5005 but nothing. I have a Dell XPS15.

thanks in advance.

closed 2016-08-01 by mh4x0f

linuggx, 2016-08-01:
Hello, thanks for the answer.
I tried to log off from my WiFi network and after I tried to start the tool, but does not work.
If I'm not connected to a WiFi network WiFi-Pumpkin shows me another error: not find internet connection.
If I start the tool without connection and after i tryed to connect to wifi network WiFi-Pumpkin shows me always this error: an connection with wlan0 has been detected : device or resource busy.

linuggx, 2016-08-01:
Hello thanks for the reply: 
if Wi-Fi Pumpkins show me this error: an connection with wlan0 has been detected : device or resource busy it means that I will never use the tool with wifi network? 
why does this happen?
Thanks

mh4x0f, 2016-08-01:
if you had a minimum knowledge of wireless networks, you will notice that it is not possible to use a  interface/wireless two ways at the same time.  you got this error because you didn't read the wiki page, please if you not user linux we recommend to use on Virtual Machine. 
thank's

Rootedskd, 2019-03-08:
Same problem is happening with me

malwareoverflow, 2020-01-26:
I had the same problem , I was using 4G device then I connected the device with laptop using it data cable (In case of router use ethernet) then turned off the laptop wifi so that the I was able to connect with internet using ethernet and built in wifi was free , just then closed the wifi pumpkin and restart run perfectly