P0cL4bs/WiFi-Pumpkin-deprecated #76: [Feature Request] Request Rogue AP, Deauth Attack and Probe Request

zero77, 2016-08-26:
WiFi-Pumpkin is a really great tool with some great features, but to make it even more powerful can you use the Rogue AP, Deauth Attack and the Probe Request together.

For example, the Probe Request picks up an AP name and then the Rogue AP creates an AP with that name and then the Deauth Attack attacks the real AP, but this is all dune automatically when a new Probe Request is captured.

Thanks

mh4x0f, 2016-08-26:
@zero77 i think that this feature does not fit in this project proposal. look this project  [wifiphisher](https://github.com/sophron/wifiphisher) , I think it would fit better... thanks

zero77, 2016-08-26:
Perhaps, but WiFi-Pumpkin is a better peas of software and it would improve it even more to automate what’s already built in.

closed 2016-08-27 by mh4x0f