P0cL4bs/WiFi-Pumpkin-deprecated #77: Using DBF Proxy No metasploit

Cosmic4x2, 2016-08-27:
Using DBF Proxy with all setting correct no payload is being sent to target followed your video using putty. also getting this error: AP::[PumpAP] Running...
AP::BSSID::[00:c0:ca:84:49:32] CH 11
Traceback (most recent call last):
  File "/root/Desktop/WiFi-Pumpkin/Core/Main.py", line 1044, in get_bdfproxy_output
    self.dockAreaList['::bdfproxy:: '].writeModeData(str(data).split(' : ')[1])
IndexError: list index out of range

[image]

mh4x0f, 2016-08-27:
try again,  send me output if it worked...

Cosmic4x2, 2016-08-27:
did not work.

mh4x0f, 2016-08-27:
send me screenshot of terminal

Cosmic4x2, 2016-08-27:
[image: Inline image 1]

mh4x0f, 2016-08-27:
I checking this

mh4x0f, 2016-08-27:
hi, everything ok with BDFproxy. I did some test in my Lab the problem is putty.org the website when started download putty.exe the request come from https:the.earth.li/~sgtatham/putty/latest/x86/putty.exe the BDFproxy not intercept HTTPS request , only  request HTTP (text plain). thank's for issue the commit above solved the bug... :D

mh4x0f, 2016-08-27:
you can you try redownloading it, but in this section the putty downloadyou can you try  again redownloading it, but in this section bellow. this  Putty listed is text plain (HTTP protocol).
[image]

Cosmic4x2, 2016-08-27:
yes it worked...

Thx

closed 2016-08-27 by mh4x0f