P0cL4bs/WiFi-Pumpkin-deprecated #83: AP SSID broadcast Failure

stellarboom, 2016-09-02:
Hi , 
Thank you for an amazing suite automating lot of things .
I have been experiencing strange problem with the tool usage. I have alfa card , packet injection is working , tested it using aircrack deauth attack, AP creation working , tested using airbase from aircrack suite..
However when i am trying to create AP through wifi-pumpkin , i can see in the console everything loads up correctly and it even says AP running with BSSID mentioned but i don't see any AP broadcast in the air.. I don't see any newly created AP with which i can connect to .. is there something i need to do in the settings which talks about ignore_ssid _broadcast...? or is it a known issue ?  Please take a look at the screenshot as well..
[image]

mh4x0f, 2016-09-07:
> AP creation working , tested using airbase from aircrack suite..

this does not mean that your card has support AP mode, the AP create from airbase-ng does not work very well for Rouge AP, the better solution is use Hostapd (WiFi-Pumpki use this for create AP). 

>  i can see in the console everything loads up correctly and it even says AP running with BSSID 
mentioned but i don't see any AP broadcast in the air

check if Card adapter support AP mode read [tutorial](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Installation#wireless-adapters), if your Card support mode AP, answer these questions below.
#### Please tell us details about your environment.
* Virtual Machine (yes or no and which): 
* Operating System and version:

wh0watme, 2016-09-08:
Greetings, I too seem to be having the same problem. I currently have the Alfa awus036h using the realtek 8187 chipset. I've been having the same problem where there is no AP in the air. It works on airbase. I've yet to try it on Hostpd. Since I can only see it on pumpkin does this mean my card does not support AP?

closed 2016-09-10 by mh4x0f