P0cL4bs/WiFi-Pumpkin-deprecated #84: Is it possible to use bettercap with fake ap enabled?

Cosmic4x2, 2016-09-18:
## What's the problem (or question)?

Is it possible to use bettercap with fake ap enabled?

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (support AP/monitor mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

mh4x0f, 2016-09-18:
yes, it's possible. just open new terminal and run bettercap (using -I options ) with same  the interface (wlanX) that you set for mount AP in **WiFi-Pumpkin**. 
ex: 
if you are using interface (wlan1) for mount AP in WiFi-Pumpkin then you need to specify this interface in **bettercap** check-out ...

>  sudo bettercap -I wlan1

Cosmic4x2, 2016-09-18:
Works Great Thx...

closed 2016-09-18 by mh4x0f

mh4x0f, 2016-10-27:
#106 look