P0cL4bs/WiFi-Pumpkin-deprecated #86: module' object has no attribute 'gateways'

Nicknam3, 2016-09-22:
## What's the problem (or question)?

Error during the execution:

```
Loading GUI...
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 52, in <module>
    app = Initialize()
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 89, in __init__
    self.form_widget    = WifiPumpkin(self,self,self.FSettings)
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 154, in __init__
    self.UpdateSoftware = frm_githubUpdate(version)
  File "/usr/share/WiFi-Pumpkin/Core/helpers/update.py", line 31, in __init__
    super(frm_githubUpdate, self).__init__(parent)
  File "/usr/share/WiFi-Pumpkin/Core/loaders/Stealth/PackagesUI.py", line 17, in __init__
    self.interfaces     = Refactor.get_interfaces()
  File "/usr/share/WiFi-Pumpkin/Core/Utils.py", line 230, in get_interfaces
    interfaces['gateway'] = netifaces.gateways()['default'][netifaces.AF_INET][0]
AttributeError: 'module' object has no attribute 'gateways'
```

#### Please tell us details about your environment.
* Card wireless adapters name (support AP/monitor mode): 
* Version used tool: last
* Virtual Machine (yes or no and which): no
* Operating System and version: Ubuntu 14.04

mh4x0f, 2016-09-22:
The `gateways() ` is only available since netifaces **0.10.0.** you have do update netifaces module.
``` sh
 sudo pip install netifaces --upgrade
```

Nicknam3, 2016-09-23:
Ok it's work with netifaces 0.10.0 :) thx and great works !

closed 2016-09-23 by Nicknam3