P0cL4bs/WiFi-Pumpkin-deprecated #87: wifi-pumpkin [✘] hostapd is not installed.

maiksir, 2016-09-25:
## What's the problem (or question)?

Hi, I've recently downloaded and tried Wifi-Pumpkin on my USB bootable KALI LINUX 2.0 Live, And It was so much fun like the 13 year olds hacking dreams haha, And then I installed kali as my main OS, And I can't seem to be able to configure wifi-pumpkin now due to hostapd not being installed..

"wifi-pumpkin
[✘] hostapd is not installed.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 50, in <module>
    from Core.Main import Initialize
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 53, in <module>
    from netfilterqueue import NetfilterQueue
ImportError: No module named netfilterqueue
"

#### Please tell us details about your environment.
* Card wireless adapters name (support AP/monitor mode): - My wireless adapter is compatible and worked the first time.

* Version used tool: I just used " git clone https://github.com/P0cL4bs/WiFi-Pumpkin.git
 cd WiFi-Pumpkin
 chmod +x installer.sh
 ./installer.sh --install

* Virtual Machine (yes or no and which): no

* Operating System and version: kali linux 2.0 (upgraded and updated)

Thanks for your time!

androuxx, 2016-09-26:
try `apt-get install hostapd` 
the will be fix your problem

mh4x0f, 2016-09-26:
you need to install hostapd ( as @androuxx  comment) and module netfilterqueue
``` sh
pip install netfilterqueue
```

closed 2016-09-26 by mh4x0f

maiksir, 2016-09-27:
Hello Guys, And thanks for taking the time to reply to my issue!

### apt-get install hostapd - 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package hostapd

And 

## pip install netfilterqueue - 

 Failed building wheel for netfilterqueue

Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-wvxniA/netfilterqueue/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-78I0Iu-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-wvxniA/netfilterqueue/
Hope to resolve this issue..

Thank you again for your time!

mh4x0f, 2016-09-27:
@maiksir search on google, Google is your friend!

maiksir, 2016-09-27:
@mh4x0f Well, I wouldn't want to bother you if I didn't search it, But it's probably some problem from my end right? 

I will uninstall and reinstall everything, And maybe this time it will all be installed correctly.

Thanks for your time once again!

mh4x0f, 2016-09-27:
@maiksir I think something is quite wrong with your sources lists. Read [here](http://docs.kali.org/general-use/kali-linux-sources-list-repositories)

maiksir, 2016-09-27:
@mh4x0f But of course!

I have it running now! 

Thanks again for your time and effort! and I hope this will help other clueless newbies like me ^^

Best Regards.