P0cL4bs/WiFi-Pumpkin-deprecated #88: Using Two WiFi Adapters with Rolling Kali

CamelToeJoe, 2016-09-25:
Rolling Kali
NO VM running as a hard disk install

Hello I have updated the NetworkManager.conf file as listed below:

    nano /etc/NetworkManager/NetworkManager.conf

[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=mac:00:xx:xx:xx:xx:xx

I have deleted the lines from Main.py below:

        if dist()[0] == 'Kali':
            if str(self.interfacesLink['activated']).startswith('wl'):
                return QMessageBox.information(self,'Error network card',
                    'You are connected with interface wireless, try again with local connection')

I then connect to the internet with my external Wireless card and confirm it is working by using the browser.
I then fire up WFP (wifi-pumpkin) and try to setup an AP on my internal wireless card as usual and I still get the alert
"You are connected with interface wireless, try again with local connection"

Anyway with out the ability to use wireless really hampers things.  I am in a situation with no wired internet and that renders WFP useless.  It is such a great program and would even better if this limitation could get fixed.

Thanks

Nicknam3, 2016-09-26:
I've the same issue here. I want to use wlan0 (internal card) to connect to Internet (with my phone as a Wi-Fi Hotspot) and a second physical Wifi adapter (alfa) to create the fake AP and make all stuff provided by wifi-pumpkin.

Nicknam3, 2016-09-26:
Just to edit my case. It's necessary to use _installer.sh --install_ after modifying _Main.py_ file.
But now that i've no error when i launch "Start Access Point", there is no "PumpAP" that showed up on Wifi available.

_ps aux | grep hostapd_ return nothing

Nicknam3, 2016-09-26:
Update: it works with my ubuntu... Maybe there is some specific issues on Kali ?

CamelToeJoe, 2016-09-26:
I believe there is a Kali specific issue.

mh4x0f, 2016-09-26:
the tool don't was tested in Kali Rolling.

closed 2016-09-26 by mh4x0f

CamelToeJoe, 2016-09-26:
Bummer.... maybe in the future.   Hope your bitcoin donation made it thru.

mh4x0f, 2016-09-26:
@CamelToeJoe 
> I then connect to the internet with my external Wireless card and confirm it is working by using the browser.
I then fire up WFP (wifi-pumpkin) and try to setup an AP on my internal wireless card as usual and I still get the alert
"You are connected with interface wireless, try again with local connection"

it makes no sense, if you are deleted the lines in code, it impossible to get this error above.

> Anyway with out the ability to use wireless really hampers things. I am in a situation with no wired internet and that renders WFP useless. It is such a great program and would even better if this limitation could get fixed.

well, please add the issue the information default for that we can analise this question. I don't know if you card internal has been support AP mode. this ploblem is not the Tool.

CamelToeJoe, 2016-09-26:
I know it doesn't make sense.  I scoured the code and couldn't find any reason to get that message nor could I find that message in the code.   

My internal card does support AP mode since I can easily use WFP when connected to the internet with a hard wire.

I will look some more info the following days.