P0cL4bs/WiFi-Pumpkin-deprecated #89: No Wifi Access Point ubuntu 16.04

Nicknam3, 2016-09-27:
## What's the problem (or question)?
When i try the tool with ubuntu 16.04, i've no error output but there is no "PumpAP" (default name) that showed up on any client devices.

My card support AP/monitor mode and on another same computer model, there is no problem with Ubuntu 14.04.

I think that hostapd is not launched by wifi-pumpkin code on Ubuntu 16.04 because i've no trace of hostapd process on ps aux output. The name of WiFi and Ethernet interface is different from Ubuntu 14.04 (it's not eth0 but eno1 and not wlan0 but wlp2s0). Maybe the problem is here ?

#### Please tell us details about your environment.
* Card wireless adapters name (support AP/monitor mode): Yes
* Version used tool: latest
* Virtual Machine (yes or no and which): no 
* Operating System and version: ubuntu 16.04

Thanks for your help.

mh4x0f, 2016-09-27:
the interface name makes no difference, because if you see card in Network Adapter option,it not have problem, the tool detect automatically. send me  the file Hostapd log in Logs/AccessPoint/hostapd.log .

Nicknam3, 2016-09-27:
Sorry, i've reinstall ubuntu 14.04 so it's working witouht problem. I think i forgot some package on ubuntu 16.04.

I can try again in 2 days with ubuntu 16.04. Can you let the bug open and i reply soon ?

Thanks again for the great works !

mh4x0f, 2016-09-27:
yes, no problem.

mh4x0f, 2016-10-16:
@Nicknam3 I did some test in Ubuntu 16.04, same problem with Parrot 3.0. I'm work for fix this. but you can mount AP using this config bellow.

you have that exclude wifi card adapter the Network-manager for do this, open WiFI-Pumpkin and go to  menu Modules->Settings. now you have to check this option and save as can see in screenshot bellow. after that, you need to **start** AP again, if doesn't work just click in **stop** and start again AP. 

OBS: the first start AP is to add the exclude interface, when start again the AP is mounted with success. I'll fixed this bug coming soon.

[image]

closed 2016-10-27 by mh4x0f

jeretc, 2016-12-04:
hi using wicd solved the issue.