P0cL4bs/WiFi-Pumpkin-deprecated #90: dsnspoof not working for Accsess Point clients

userdsp, 2016-09-28:
loading pages just stop. Im using SetEnv path

* Card wireless adapters name (support AP/monitor mode): TL-WN722N
* Version used tool: 0.8.1
* Virtual Machine (yes or no and which): yes
* Operating System and version: Kali Linux 4.6.0-kali1-686-pae

mh4x0f, 2016-09-28:
I'll add a commit coming soon for fix this bug. thanks for report

mh4x0f, 2016-09-29:
@userdsp try now...

userdsp, 2016-09-29:
Still not working... If I start dnsspoof then all pages stoped loading at clients side and there is no redirecting to the phished page

mh4x0f, 2016-09-29:
send me printscreen the dnsspoof window.

mh4x0f, 2016-09-29:
@userdsp for Dnsspoof work with Access Point, you need to set plugin for "No Proxy". as can see in screenshot.
[image]

userdsp, 2016-09-29:
If I stop proxy then all pages has loaded except the spoofed one. 
[image]

mh4x0f, 2016-09-29:
It's not working  because facebook is using HTTPS. dnsspoof only work for HTTP trafic.

closed 2016-09-29 by mh4x0f