P0cL4bs/WiFi-Pumpkin-deprecated #93: No Wifi Access Point

bunzymr, 2016-10-01:
Dear Sir/ Madam,
Please be informed I faced the same issue which has been mentioned here:
"https://github.com/P0cL4bs/WiFi-Pumpkin/issues/89"
After start access point, there is no AP name appears in any devices and there is no error message.
More Info: 
OS:                                 Kali 2016 V2 64 bit
Wifi Card:                      Alfa AWUS036H with Monitor/ Packet Injection support 
Wifi-Pumpkin Version:  0.8.1
Virtual Machine:           Vmware Work Station Pro 12
FYI: ALFA card has been connected with usb cable and has been mounted in Kali linux.

lszb19, 2016-10-01:
Same here. AWUS036H don't support AP mode :(

closed 2016-10-01 by mh4x0f

bunzymr, 2016-10-15:
Strongly suggest to developers, (Specially "mh4X0f")
Instead of closing the issue without acceptable reason/s , take a look deeply in the issue.
Did not read the wiki? lol 
OK. FYI, Since that time I tried with the tested OS which are mentioned in the Wiki. Kubuntu 15.10 and Wifislax 11.
Same issue. There is no Access point visible after start the pumpkin.
Both in Vmware Workstation 12,
Both 64 Bit,
Wifi Card: AWUS036H 
Wifi-Pumpkin Ver: 0.8.3 Last Update 12/10/2016

mh4x0f, 2016-10-15:
>  acceptable reason/s ?

this issue has been explained above

@bunzymr look... this comment of  @lszb19, [here](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/93#issuecomment-250926960) if you read the Wiki page or FAQ you would see that your wifi card adapter don't support AP mode.

bunzymr, 2016-10-15:
Dear friend,
Thanks for your prompt and nice reply.