P0cL4bs/WiFi-Pumpkin-deprecated #97: FakeAP problems

Buff3r, 2016-10-13:
## What's the problem (or question)?
When I try to start FakeAP it doesn't start.

Logs:
WiFi-Pumpkin Report Logger
Report Generated at::Thu Oct 13 10:19:56 2016
-[ hostapd Logger ]-
SessionID[OTEyNzg=] 2016-10-13 09:49:49,590 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[OTEyNzg=] 2016-10-13 09:50:32,828 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[MjA2NDQ=] 2016-10-13 09:50:32,828 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[OTEyNzg=] 2016-10-13 09:53:57,905 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[MjA2NDQ=] 2016-10-13 09:53:57,905 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[NzEyNzA=] 2016-10-13 09:53:57,905 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[NDc4MDg=] 2016-10-13 10:05:23,914 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

SessionID[NDc4MDg=] 2016-10-13 10:08:43,112 : Configuration file: Settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
[... 196 lines cut ...]
SessionID[ODQ0NA==] 2016-10-13 10:09:30,743 : Server starting service.
SessionID[NzA2MzA=] 2016-10-13 10:09:30,743 : Server starting service.
SessionID[MTE0MDg=] 2016-10-13 10:17:21,572 : ---[ Start DHCP Thu Oct 13 10:17:21 2016]---
SessionID[MTE0MDg=] 2016-10-13 10:17:21,575 : Internet Systems Consortium DHCP Server 4.3.5b1
SessionID[MTE0MDg=] 2016-10-13 10:17:21,575 : Copyright 2004-2016 Internet Systems Consortium.
SessionID[MTE0MDg=] 2016-10-13 10:17:21,578 : All rights reserved.
SessionID[MTE0MDg=] 2016-10-13 10:17:21,578 : For info, please visit https://www.isc.org/software/dhcp/
SessionID[MTE0MDg=] 2016-10-13 10:17:21,588 : Config file: /etc/dhcp/dhcpd.conf
SessionID[MTE0MDg=] 2016-10-13 10:17:21,588 : Database file: /var/lib/dhcp/dhcpd.leases
SessionID[MTE0MDg=] 2016-10-13 10:17:21,589 : PID file: /var/run/dhcpd.pid
SessionID[MTE0MDg=] 2016-10-13 10:17:21,589 : There's already a DHCP server running.
SessionID[MTE0MDg=] 2016-10-13 10:17:21,589 : 
SessionID[MTE0MDg=] 2016-10-13 10:17:21,589 : If you think you have received this message due to a bug rather
SessionID[MTE0MDg=] 2016-10-13 10:17:21,589 : than a configuration issue please read the section on submitting
SessionID[MTE0MDg=] 2016-10-13 10:17:21,590 : bugs on either our web page at www.isc.org or in the README file
SessionID[MTE0MDg=] 2016-10-13 10:17:21,591 : before submitting a bug.  These pages explain the proper
SessionID[MTE0MDg=] 2016-10-13 10:17:21,591 : process and the information we find helpful for debugging..
SessionID[MTE0MDg=] 2016-10-13 10:17:21,591 : 
SessionID[MTE0MDg=] 2016-10-13 10:17:21,591 : exiting.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TL-WN722N
* Version used tool: last
* Virtual Machine (yes or no and which): no
* Operating System and version: Kali Linux 2016.2 (Rolling)

mh4x0f, 2016-10-13:
the tool don't was tested in Kali Rolling.

closed 2016-10-13 by mh4x0f

mh4x0f, 2016-10-13:
@Buff3r you can try to exclude card the network-manager and check if work. [here tutorial](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/69#issuecomment-245456151)