P0cL4bs/WiFi-Pumpkin-deprecated #98: Fake AP Doesnt lunch on ubuntt 16 when using 2 wireless cards

ceromee, 2016-10-15:
## What's the problem (or question)?
I have installed ubuntu 16, and I connect one wireless card to the internet, and I launch the fake ap using an usb wireless card, and I got not errors, but the ap doesnt get launch
*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
I am using:
*Atheros AR3012(supports monitor mode)
*AWUS036H(supports monitor mode)

* Version used tool: 0.8.1

* Virtual Machine (yes or no and which): no

* Operating System and version: Ubuntu 16.04

mh4x0f, 2016-10-15:
@ceromee you need to update tool....
you have that exclude wifi card adapter the Network-manager for do this, open WiFI-Pumpkin and go to  menu Modules->Settings. now you have to check this option and save as can see in screenshot bellow. after that, you need to **start** AP again, if doesn't work just click in **stop** and start again AP. 

OBS: the first start AP is to add the exclude interface, when start again the AP is mounted with success. I'll fixed this bug coming soon.

[image]

mh4x0f, 2016-10-15:
this card not have support AP mode

closed 2016-10-15 by mh4x0f