P0cL4bs/WiFi-Pumpkin-deprecated #101: SSL strip. Exception raised

vsegdacocacola, 2016-10-21:
## What's the problem (or question)?

Hi!

When i run AP with SSLstrip Proxy option on, i have exception (see below) on any HTTPs site open attempt.

Unhandled Error
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/Core/utility/threads.py", line 221, in run
    reactor.run(installSignalHandlers=False)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1194, in run
    self.mainLoop()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1206, in mainLoop
    self.doIteration(t)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 396, in doPoll
    log.callWithLogger(selectable, _drdw, selectable, fd, event)
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 101, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 84, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 610, in _doReadOrWrite
    self._disconnectSelectable(selectable, why, inRead)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 255, in _disconnectSelectable
    selectable.connectionLost(f)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 479, in connectionLost
    self._commonConnection.connectionLost(self, reason)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 293, in connectionLost
    protocol.connectionLost(reason)
  File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 486, in connectionLost
    self.handleResponseEnd()
  File "/usr/share/WiFi-Pumpkin/Plugins/sslstrip/ServerConnection.py", line 133, in handleResponseEnd
    HTTPClient.handleResponseEnd(self)
  File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 497, in handleResponseEnd
    self.handleResponse(b)
  File "/usr/share/WiFi-Pumpkin/Plugins/sslstrip/ServerConnection.py", line 163, in handleResponse
    self.client.write(data)
  File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 896, in write
    raise RuntimeError('Request.write called on a request after '
exceptions.RuntimeError: Request.write called on a request after Request.finish was called.

#### Please tell us details about your environment.
* Card wireless adapters name: TP-LINK TL-WN722N
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): Parallels VM
* Operating System and version: Kali 2.0

mh4x0f, 2016-10-22:
This an #57 were the same.

closed 2016-10-22 by mh4x0f

alalamin25, 2017-11-27:
I have the same configuration and having same error