P0cL4bs/WiFi-Pumpkin-deprecated #102: Bug?

diego-linares-boquete, 2016-10-23:
Hello
I have had this error, but if you create the folder manually, and Reloading the program, the logs are generated normally and everything continues well. I guess some will be writing problem
/////////////////////

root@kali:/opt/Wifi-Pumpkin/WiFi-Pumpkin# sudo python wifi-pumpkin.py 
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [NDMyNTk=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
Traceback (most recent call last):
  File "/opt/Wifi-Pumpkin/WiFi-Pumpkin/core/main.py", line 1219, in Start_PumpAP
    thread.start()
  File "/opt/Wifi-Pumpkin/WiFi-Pumpkin/core/utility/threads.py", line 198, in start
    self.makeLogger()
  File "/opt/Wifi-Pumpkin/WiFi-Pumpkin/core/utility/threads.py", line 206, in makeLogger
    setup_logger('hostapd', './logs/AccessPoint/hostapd.log',self.session)
  File "/opt/Wifi-Pumpkin/WiFi-Pumpkin/core/utils.py", line 90, in setup_logger
    fileHandler = logging.FileHandler(log_file, mode='a')
  File "/usr/lib/python2.7/logging/__init__.py", line 913, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 943, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 2] No such file or directory: '/opt/Wifi-Pumpkin/WiFi-Pumpkin/logs/AccessPoint/hostapd.log'
killing all threads...
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[DHCP] successfully stopped.
root@kali:/opt/Wifi-Pumpkin/WiFi-Pumpkin# sudo python wifi-pumpkin.py 
Loading GUI...
////////

Thanks for developing

mh4x0f, 2016-10-23:
this is not an bug writing problem, I check in code, it's all OK. you  downloaded the tool is using old version directory. for this the folder "Logs" is not found because now is "logs" folder. my suggestion is reinstall download again the tool project and reinstall with "installer.sh" will be solve problem.

mh4x0f, 2016-10-26:
@diego-linares-boquete I'm sorry, I'll solved this bug coming soon

closed 2016-10-26 by mh4x0f