P0cL4bs/WiFi-Pumpkin-deprecated #104: wifi-pumpkin not work

arbalest722, 2016-10-26:
root@kali:~/WiFi-Pumpkin# wifi-pumpkin 
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [NDQzMzE=]
[*] Configuring hostapd...
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/Core/Main.py", line 1102, in Start_PumpAP
    return QMessageBox.warning(self,'Error nmcli',e)
TypeError: arguments did not match any overloaded call:
  QMessageBox.warning(QWidget, QString, QString, QMessageBox.StandardButtons buttons=QMessageBox.Ok, QMessageBox.StandardButton defaultButton=QMessageBox.NoButton): argument 3 has unexpected type 'exceptions.OSError'
  QMessageBox.warning(QWidget, QString, QString, int, int, int button2=0): argument 3 has unexpected type 'exceptions.OSError'
  QMessageBox.warning(QWidget, QString, QString, QString, QString button1Text=QString(), QString button2Text=QString(), int defaultButtonNumber=0, int escapeButtonNumber=-1): argument 3 has unexpected type 'exceptions.OSError'

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):support
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): yes,virtualbox
* Operating System and version: kali 2.0

mh4x0f, 2016-10-26:
I'll add some modifications to the code will correct the error but you will receive an error when run AP because "nmcli" is not installed in kali, I recommend researching how to install it.

arbalest722, 2016-10-27:
Thankyou very much @mh4x0f. For your advice, i find something wrong with my  network-manager service. I solved the problem when i reinstall zhe service.

closed 2016-10-27 by mh4x0f