P0cL4bs/WiFi-Pumpkin-deprecated #106: SSLstrip

jeretc, 2016-10-27:
## What's the problem (or question)?

*hi, does the sslstrip work? i can only get the credentials for non https sites. i have enabled the plugins. When i use bettercap it can strip certain sites like hotmail.com, mail.com by adding "wwww". does wifi-pumpkin does the same?

Also like to comment that the gui interface of wifi-pumpkin is really nice and well done*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TL-WN722N
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): no
* Operating System and version: Parrot Security Os

mh4x0f, 2016-10-27:
hi, the sslstrip not work for all website, because HSTS implementation,if you want to use WiFi-Pumpkin + Bettercap you need to disable plugins for this you should go in to Plugins tab and uncheck the option Enable Proxy Server as can see in screenshot bellow.
[image]

jeretc, 2016-10-27:
Ok , thanks for the info.

closed 2016-10-27 by jeretc