P0cL4bs/WiFi-Pumpkin-deprecated #108: Raspberry Pi 3 + ALFA036NH = Tracebacks

GGGunrunner, 2016-11-01:
Hi everybody!

Playing with WIFI-Pumpkin now. The setup is:

Pi 3 with Ubuntu Mate
Alfa 036NH

Surprisingly, it has installed without any errors (while 99% of github projects wont :)
Only had to install python's openssl lib and service_identity modules (a check for them you might like to add in the future?)

The PumpAP is working fine, my smartphones can get online. However, it throws alot of errors.
Could it be due to Pumpkin's low speed in analyzing the traffic?

Here's the output:
====

Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [NTEyMDk=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 11405 (hostapd)]
[New Thread 11417 (DHCP)]
[New Thread 11420 (Net-Creds)]
[New Thread 11433 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve
-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[00:c0:ca:64:de:25] CH 10
Unhandled Error
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 221, in run
    reactor.run(installSignalHandlers=False)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/base.py", line 1195, in run
    self.mainLoop()
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/base.py", line 1207, in mainLoop
    self.doIteration(t)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 235, in doPoll
    log.callWithLogger(selectable, _drdw, selectable, fd, event)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/log.py", line 101, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/log.py", line 84, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 610, in _doReadOrWrite
    self._disconnectSelectable(selectable, why, inRead)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 255, in _disconnectSelectable
    selectable.connectionLost(f)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 478, in connectionLost
    self._commonConnection.connectionLost(self, reason)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 292, in connectionLost
    protocol.connectionLost(reason)
  File "/usr/local/lib/python2.7/dist-packages/twisted/web/http.py", line 486, in connectionLost
    self.handleResponseEnd()
  File "/usr/share/WiFi-Pumpkin/plugins/sslstrip/ServerConnection.py", line 133, in handleResponseEnd
    HTTPClient.handleResponseEnd(self)
  File "/usr/local/lib/python2.7/dist-packages/twisted/web/http.py", line 497, in handleResponseEnd
    self.handleResponse(b)
  File "/usr/share/WiFi-Pumpkin/plugins/sslstrip/ServerConnection.py", line 163, in handleResponse
    self.client.write(data)
  File "/usr/local/lib/python2.7/dist-packages/twisted/web/http.py", line 896, in write
    raise RuntimeError('Request.write called on a request after '
exceptions.RuntimeError: Request.write called on a request after Request.finish was called.

mh4x0f, 2016-11-01:
This an #101 were the same.

closed 2016-11-01 by mh4x0f

mh4x0f, 2016-11-01:
> Surprisingly, it has installed without any errors (while 99% of github projects wont :)
Only had to install python's openssl lib and service_identity modules (a check for them you might like to add in the future?)

@GGGunrunner this modules  is not necessary for the systems supported by the tool.