P0cL4bs/WiFi-Pumpkin-deprecated #112: ERRO: File "/usr/share/WiFi-Pumpkin/core/main.py", line 1093, in Start_PumpAP print '[!] ' + e

alexandrebfaust, 2016-11-09:
## What's the problem (or question)?

Help?!

```
pi@raspberrypi:~ $ sudo wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [Mzc4OTM=]
[*] Configuring hostapd...
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1093, in Start_PumpAP
    print '[!] ' + e
TypeError: cannot concatenate 'str' and 'exceptions.OSError' objects
```

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TL-WN722N - TP-Link
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): Nope
* Operating System and version:

mh4x0f, 2016-11-09:
fixed, but you will continue to get an error from nmcli, my suggestion is install it.

alexandrebfaust, 2016-11-09:
Whats the pack to install nmcli?

mh4x0f, 2016-11-09:
the nmcli is command tool the network-manager 

> sudo apt-get install network-manager

alexandrebfaust, 2016-11-09:
Solved! Thanks ;)

alexandrebfaust, 2016-11-09:
Now this :/
[image]

mh4x0f, 2016-11-09:
look, Before send error look in issue closed #57

closed 2016-11-09 by mh4x0f