P0cL4bs/WiFi-Pumpkin-deprecated #116: 0.8.3 Issue

nikithesmat, 2016-11-23:
No internet for the client when using SSL+Dns2
when im trying to create a SSID with no proxy or sergio proxy,Wifi-Pumpkin closes itself with this error :
AP: :BSSID: :[ MAC ADRESS ] CH 10
/usr/bin/wifi-pumpkin: line2: 3098 Killed                               sudo python wifi-pumpkin.py

AWUS051NH as device to share WiFi and using bridged connection from host
V0.8.3
VMWare 11 with Kali linux 2016.2 with yesterday apt update
Operating System Windows 10 64Bit   8Gb Ram   Geforce 940MX  and  i5 7th Gen

what can be the issue ?

mh4x0f, 2016-11-23:
nice issue, I'll add new commit for fix this bug thanks :D

mh4x0f, 2016-11-23:
if you got "No internet for the client " again, maybe the bridged connection is the problem. try with NAT connection.

nikithesmat, 2016-11-23:
Sounds good thanks chief

nikithesmat, 2016-11-23:
I Just tried and still no changes,i also did the update for the fix but no work but i cant acess the internet with the Fake Ap and also switched to NAT connection.

mh4x0f, 2016-11-23:
this is very strange, but try change in Settings-Genera ->  DNSServer to "isc dhcp server (dhcpd)" option and if you got "dhcpd is not installed" type in terminal this command: 
``` sh 
sudo apt-get install isc-dhcp-server
``` 
after that you can to run again the AP.

nikithesmat, 2016-11-23:
Works good now. Also im having issue with the tables i got 2 devices connected but it says 0 and none listed.

nikithesmat, 2016-11-23:
Why when lets say i want to detect creds from audizine.com , i get the username but the password is full of codes and not the password i have use in ?

mh4x0f, 2016-11-23:
1-  the table get the info from dhcpd, if dhcpd is not detect device then you can't to see. 
2- probably, the audizine encrypt the password after the send to server.

closed 2016-11-23 by mh4x0f

nikithesmat, 2016-11-23:
You cannot do a way like we get the password that has been typed originally ?
Most bank sites wont work too i think right ?
Can u do a video of injection-proxy more detailed ? like any device that connects automaticly downloads your rat.exe silently or something like this ?