P0cL4bs/WiFi-Pumpkin-deprecated #117: wifi-pumpkin start up error

jeretc, 2016-11-24:
## What's the problem (or question)?

*WARNING: Failed to execute tcpdump. Check it is installed and in the PATH
:0: UserWarning: You do not have a working installation of the service_identity module: 'No module named service_identity'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module and a recent enough pyOpenSSL to support it, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Loading GUI...
WiFi-Pumpkin Running!

i have openssl and service_identity installed.
*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):722N
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): no
* Operating System and version: parrot 3.2

mh4x0f, 2016-11-26:
install pyOpenSSL:
``` sh
$ sudo easy_install PyOpenSSL
```

closed 2016-11-29 by mh4x0f

KaliMalibu, 2018-03-31:
Hi,

I am getting same error, and try to do
```

root@kali:~# sudo easy_install PyOpenSSL
Searching for PyOpenSSL
Best match: pyOpenSSL 17.5.0
Adding pyOpenSSL 17.5.0 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Processing dependencies for PyOpenSSL
Finished processing dependencies for 

PyOpenSSL
```

but error is still there.
How can I fix that? Thank you for yours help.